lvgl/lv_esp_idf #3: disp_spi driver fix?

MichaelVLV, 2019-02-05:
Hi.

I'm using this example with another kind of spi display, but in general, all works without much changes, only command sequence need to adjust.

With latest spi driver nothing dispays on screen. 

I'm able to fix it if added:

_**spi_device_get_trans_result()**_ to both functions _disp_spi_send_data()_ and _disp_spi_send_colors()_

So it looks like next now
```C
void disp_spi_send_data(uint8_t * data, uint16_t length)
{
    if (length == 0) return;           //no need to send anything

    while(spi_trans_in_progress);

    spi_transaction_t t;
    memset(&t, 0, sizeof(t));       	//Zero out the transaction
    t.length = length * 8;              //Length is in bytes, transaction length is in bits.
    t.tx_buffer = data;               	//Data
    spi_trans_in_progress = true;
    spi_color_sent = false;             //Mark the "lv_flush_ready" NOT needs to be called in "spi_ready"
    spi_device_queue_trans(spi, &t, portMAX_DELAY);

    //added
    spi_transaction_t * rt;
    spi_device_get_trans_result(spi, &rt, portMAX_DELAY);
}

void disp_spi_send_colors(uint8_t * data, uint16_t length)
{
    if (length == 0) return;           //no need to send anything

    while(spi_trans_in_progress);

    spi_transaction_t t;
    memset(&t, 0, sizeof(t));           //Zero out the transaction
    t.length = length * 8;              //Length is in bytes, transaction length is in bits.
    t.tx_buffer = data;                 //Data
    spi_trans_in_progress = true;
    spi_color_sent = true;              //Mark the "lv_flush_ready" needs to be called in "spi_ready"
    spi_device_queue_trans(spi, &t, portMAX_DELAY);

    //added
    spi_transaction_t * rt;
    spi_device_get_trans_result(spi, &rt, portMAX_DELAY);
}
```

kisvegabor, 2019-02-06:
I updated the driver to remove `spi_device_get_trans_result` because it means a blocking wait. Now I can enable `LV_VDB_DOUBLE` to make rendering a flushing parallel. This way achieved 30 FPS on 320 x 240 display. See this [video](https://www.youtube.com/watch?v=EiIHI8sE7as). So it would be important to not use `spi_device_get_trans_result` but `post_cb` instead.

To see what's going on in your case:
1. Be sure you are using the latest master branch and updated lvgl to latest master too.
2. Please add some `printf`s to the SPI driver functions and post log.

closed 2019-02-06 by MichaelVLV

kisvegabor, 2019-02-07:
Was the issue solved?