lvgl/lv_esp_idf #6: touch not working

alen1359, 2019-05-08:
i am using 2.8 inch TFT 320x240 version 1.1 with xpt 2046 touch drive, but the touch part is not working, could you please advice?

i thought it was the screen and bought another one but i have the same issue

link for my screen: https://www.esp8266.com/viewtopic.php?f=160&t=14302

alen1359, 2019-05-08:
seems like the touch work but there is big offset between where i touch and it react

alen1359, 2019-05-08:
if i touch top left i get  x=320, y=239, if i touch top right i get x= 4, y=240, if i touch bottom left i get x=320, y=-416 (note: it is negative 416), if i touch bottom right i get x=2, y=-235

seems like need calibration, could you please help

kisvegabor, 2019-05-08:
It seems you just need to swap the x coordinate. You can do it easily here: https://github.com/littlevgl/esp32_ili9341/blob/master/drv/xpt2046.c#L95

```c
x = 320-x;
if( x < 0) x = 0;
if(x  > 319) x = 319;
last_x = x; 
```

alen1359, 2019-05-08:
with that correction i get the following result

if i touch top left i get x=0, y=239, if i touch top right i get x= 320, y=240, if i touch bottom left i get x=0, y=-416 (note: it is negative 416), if i touch bottom right i get x=320, y=-293

here where i added:
        /*Normalize Data*/
        x = x >> 3;
        y = y >> 3;
        xpt2046_corr(&x, &y);
        xpt2046_avg(&x, &y);
		
		x = 320-x;
		if( x < 0) x = 0;
		if(x  > 319) x = 319;
		last_x = x; 

        //last_x = x;
        last_y = y;

alen1359, 2019-05-08:
more accurate data using pen for touch

if i touch top left i get x=0, y=239, if i touch top right i get x= 320, y=240, if i touch bottom left i get x=0, y=-416 (note: it is negative 416), if i touch bottom right i get x=320, y=-416

kisvegabor, 2019-05-09:
I see. It really still need to be calibrated. 
The calibration values are here:
https://github.com/littlevgl/esp32_ili9341/blob/master/drv/xpt2046.h#L27-L32

You should try to set the MIN coordinates to 0 and MAX to 4096 to disable calibration, check the coordinates and write them as calibration values.

I forget that here you can also invert the X and/or Y coordinates if required. So the previous [hack](https://github.com/littlevgl/esp32_ili9341/issues/6#issuecomment-490345388) is not required.

alen1359, 2019-05-09:
ok, if i do min=0 and max 4096 i get teh followings

#define XPT2046_X_MIN       39
#define XPT2046_Y_MIN       21
#define XPT2046_X_MAX       271
#define XPT2046_Y_MAX       226

but when i put them in my code it doesn't work still i also tried the followings but no success

#define XPT2046_X_MIN       39
#define XPT2046_Y_MIN       21
#define XPT2046_X_MAX       271
#define XPT2046_Y_MAX       226

Note, X invert need to be 0

thanks for all your help

kisvegabor, 2019-05-10:
What was the problem with the first version?

alen1359, 2019-05-11:
ok i ran it again with 0 and 4096, here are the numbers,
bottom left 	x=38, 	y=21
top left 		x=38, 	y=211
top right 		x=250, 	y=226
bottom right	x= 272,	y=29

if i enter these numbers in my code and compile i get the followings
#define XPT2046_X_MIN 38
#define XPT2046_Y_MIN 21
#define XPT2046_X_MAX 272
#define XPT2046_Y_MAX 226

i get 

bottom left 	x=379, 	y=-4118
top left 		x=378, 	y=-333
top right 		x=4334, 	y=-33
bottom right	x= 5024,	y=-4081

which make no sense

kisvegabor, 2019-05-13:
I have no idea. The correction happens here: https://github.com/littlevgl/esp32_ili9341/blob/master/drv/xpt2046.c#L116

With the parameters, you provided it should work. You can debug it using `printtf`s to see where the miscalculation happens.

alen1359, 2019-05-15:
i can't figure out what is going on, is it possible to recommend me a display that works with this code.

thanks,

alen1359, 2019-05-15:
i set the invert for Y to 1 and i believe this is where the problem is, it is creating a negative number for Y

#if XPT2046_Y_INV != 0
    (*y) =  XPT2046_VER_RES - (*y);

alen1359, 2019-05-15:
top left should be 0,0 or bottom left ?

kisvegabor, 2019-05-15:
> top left should be 0,0 or bottom left ?

Top-left should be (0,0).

> i can't figure out what is going on

You should `printf` the raw and corrected data step by step to where the problem occurs. 

>  is it possible to recommend me a display that works with this code.

Resistive TPs might be different even for the same product type.

C47D, 2019-06-10:
I also noticed the touch isn't working, in my case when i touch the screen from right to left the red dots appear from top to bottom, when i touch the screen from bottom to top the red dots appears from right to left.

I haven't tried to log the input device output, is there a 'easy' way to do it?

Regards

embeddedt, 2019-06-10:
@C47D Use `printf` or something similar inside the `indev_read` function.

C47D, 2019-06-10:
Thanks, i will take a look at it.

closed 2019-07-08 by stale[bot]