lvgl/lv_esp_idf #8: Cannot get the example to work

agent3bood, 2019-07-04:
Steps to produce the issue:
1. follow cloning and copying component.mk files
2. make manuconfig and change serial flasher to 115200
3. I am using this display http://www.lcdwiki.com/2.8inch_SPI_Module_ILI9341_SKU:MSP2807, ESP module is https://wiki.wemos.cc/products:d32:d32
4. connect the following pins (MOSI, SCLK, CS, DC, BCKL, RST) as per default config
5. make flash monitor
6. the error is bellow
```
ILI9341 initialization.
Enable backlight.
E (11314) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (11314) task_wdt:  - IDLE0 (CPU 0)
E (11314) task_wdt: Tasks currently running:
E (11314) task_wdt: CPU 0: main
E (11314) task_wdt: CPU 1: IDLE1
```

kisvegabor, 2019-07-05:
Have you changed anything in the code?

agent3bood, 2019-07-05:
No, I didn't change any code, also I tried to copy the example to my own project and got the same error.
The screen background led will go on and nothing on the display.

kisvegabor, 2019-07-06:
I'll update the project to the `v6.0` version of LittlevGL next week. Hopefully, the problem will be solved then.
If not, at least I will have an ESP32 with a display to make further tests. 

BTW, the error message says "watchdog got triggered". Is the watchdog enabled by default?

To see what happens in LittelvGL you can enable [Logging](https://docs.littlevgl.com/en/html/porting/log.html)

agent3bood, 2019-07-06:
Thanks I will wait for the updated repo.
About the error message I think it is from FreeRTOS.
I have also enabled logging for LittelvGL, it is logging that the UI is being drown but nothing on the screen and I get the same watchdog error. 
I have tried different boards and screens and different example configurations it all give the same error. 
By the way I update the LittelvGL library, can I just pull from master?
Regards

kisvegabor, 2019-07-06:
> By the way I update the LittelvGL library, can I just pull from master?

Yes, but note that the prototype of the drivers has been changed.  See these about the [Display](https://docs.littlevgl.com/en/html/porting/display.html) and [Touchpad](https://docs.littlevgl.com/en/html/porting/indev.html#touchpad-mouse-or-any-pointer).

You should also update the [lv_example](https://github.com/littlevgl/lv_examples) repo.

C47D, 2019-07-06:
I disabled the watchdog interrupt to don't get those warnings, it's not recommended to do so but it's just for trying out lvgl.
After doing `make menuconfig` go to `Component config > Common ESP Related` and disable `Interrupt watchdog` and `Initialize Task watchdog timer on startup`.

kisvegabor, 2019-07-10:
I've updated the project ad found the issue. It was the touchpad driver.

Is it working there too?

agent3bood, 2019-07-10:
I could not get the touch to work on either this repo or https://github.com/espressif/esp-iot-solution/tree/master/examples/hmi
On the iot-solutions example I got the display to work only.

kisvegabor, 2019-07-10:
In this project, is the display working?

agent3bood, 2019-07-10:
yes display is working in  iot-solutions  https://github.com/espressif/esp-iot-solution/tree/master/examples/hmi

kisvegabor, 2019-07-10:
> yes display is working in iot-solutions 

However, it still uses  v5.3. In this issue, the goal is to update this project to v6.0

pjaos, 2019-08-04:
Not sure if you've solved this issue but I've implemented a driver for the ADS7843 touch screen driver used in this display (http://www.lcdwiki.com/2.8inch_SPI_Module_ILI9341_SKU:MSP2807), integrated this with great LittlevGL and mongoose OS. It's available at https://github.com/pjaos/mgos_esp32_littlevgl if you're interested.

agent3bood, 2019-08-06:
Hi
Actually I did not fix the issue as I was working on other parts of the project (it is part time so it is moving slow). I will definitely check the new one also and report back here is the issue is solved.
Thanks again

kisvegabor, 2019-08-06:
I suggest continuing it on the Forum because more people are reading that.
https://forum.littlevgl.com

agent3bood, 2019-08-06:
I can confirm the display example is working with latest esp idf pulled today.

closed 2019-08-06 by agent3bood

kisvegabor, 2019-08-07:
Awesome, thank you for the feedback!