lvgl/lv_esp_idf #9: lvgl as a esp32-idf component

C47D, 2019-07-07:
Hi,

Is it possible to add **lvgl** as a `component` (in the esp32 idf argon)? [here](https://github.com/C47D/esp32_lvgl_experiments) is an example on how (see components directory).

The **lvgl** directory should be added into a directory named `components`, the same goes for the **drv** directory. The `component.mk` file that goes into the **drv** directory looks like this:

```
COMPONENT_SRCDIRS := .
COMPONENT_ADD_INCLUDEDIRS := $(COMPONENT_SRCDIRS) ..
```

The `component.mk` file that goes into the **lvgl** directory looks like this:
```
COMPONENT_SRCDIRS := . \
	./src/lv_core \
	./src/lv_draw \
	./src/lv_objx \
	./src/lv_hal \
	./src/lv_misc \
	./src/lv_misc/lv_fonts \
	./src/lv_themes \
	./src/lv_fonts
COMPONENT_ADD_INCLUDEDIRS := $(COMPONENT_SRCDIRS) ..
```

kisvegabor, 2019-07-08:
Hi,

I'm not an expert with ESP32, but what is the advantage of this structure?

C47D, 2019-07-08:
Hi,

Neither i am an expert ESP32 user, but using **components** is just the recommendation of Espressif on how organize projects. See [here](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#component-makefiles).

> “components” are modular pieces of standalone code which are compiled into static libraries (.a files) and linked into an app. Some are provided by esp-idf itself, others may be sourced from other places.

And below there's a tree of an example project, and explains the `components` directory:

> This example “myProject” contains the following elements:
> Optional “components” directory contains components that are part of the project. A project does not have to contain custom components of this kind, but it can be useful for structuring reusable code or including third party components that aren’t part of ESP-IDF.

kisvegabor, 2019-07-08:
Hm, this project already uses `components.mk` see the [README](https://github.com/littlevgl/esp32_ili9341#add-littlevgl-to-the-build). So basically the update you suggest is to move the components to a `coponents` folder because it's more standard?
If so, I agree, it's better to follow the ESP's rules if possible.

C47D, 2019-07-08:
> So basically the update you suggest is to move the components to a coponents folder because it's more standard?

Yes. I found it easier to work with that way.

The component examples i uploaded in [this](https://github.com/littlevgl/esp32_ili9341/issues/9#issue-464968522) comment are just suggestions, adapted to the new directory structure of lvgl v6.0.

kisvegabor, 2019-07-09:
> Yes. I found it easier to work with that way.

I see. We can do it as part of the migration to `v6.0`. 
Hopefully, I will get back my ILI9341 screen from my friend today and can update the project.

C47D, 2019-07-09:
Great, thanks :-)

kisvegabor, 2019-07-10:
I've updated the project I couldn't make it work with the proposed directory structure yet.
Now it has the old structure but with lvgl 6.0.

Can you take look at it?

C47D, 2019-07-10:
Great!, yes, i can take a look at it.

C47D, 2019-07-11:
Hi,

I got it working, i also had to move the `lv_conf.h` and `lv_ex_conf.h` files into the `components` directory. I will have to update the `git submodules` path and expect the PR soon.

kisvegabor, 2019-07-11:
Great! Waiting for the PR :)

C47D, 2019-07-11:
Let me know if like the new README

PR #10 (2019-07-15): Use Espressif recommended directory structure for ESP32 projects
Fixes #9 and also updates the README. Please test it and let me know what do you think.

Carlos
commit e2471ba5 (2019-07-15): Move lvgl and lv_examples submodules to the components directory as recommended by Espressif, fixes #9
closed 2019-07-15 by kisvegabor