lvgl/lv_esp_idf #11: ili9341 driver backlight

Erol2309, 2019-07-11:
Hello,

In `ili9341.c` i think there might be a mistake in the `ili9341_init()` function. I changed the last instruction to `gpio_set_level(ILI9341_BCKL, 0)` as opposed to `gpio_set_level(ILI9341_BCKL, 1)`. This might be due to the pull-up or pull-down config of the gpio but i am not sure.
Posted this for anybody that might encounter this.

kisvegabor, 2019-07-12:
Hi,

It should be specific to the display. Probably your display needs '0' on the backlight pin to enable it.

C47D, 2019-07-13:
Hi,

I'm Carlos at the forum and found the same issue [1], maybe we can use some symbols and build the project like so: `make CFLAGS += -DESP32_BOARD_WROVER_KIT`, this will make the following snippet set the correct level of the BCLK pin.

```c
#if defined (ESP32_BOARD_WROVER_KIT)
gpio_set_level(ILI9341_BCKL, 0)
#else // Other boards
gpio_set_level(ILI9341_BCKL, 1)
#endif
```

This can also help the users to assign the right pin out of their boards without modifying any file, only setting the according symbol as above.

What do you think?

[1] [Lvgl in ESP-WROVER-KIT v4.1](https://forum.littlevgl.com/t/lvgl-in-esp-wrover-kit-v4-1/226)

kisvegabor, 2019-07-15:
Hi Carlos,

Sounds good! 
If there is no obsctacle we can use the Wrover Kit's pins as default. On Wroom32 modules, we can use any pins so we can be compatible with the Wroomkit by default.
Maybe the TP's SPI could be an issue because I saw that you need to use the SPI in half-duplex mode to achieve 40 MHz.

C47D, 2019-07-15:
The Wrover kit doesn't have a touch screen, it's just the TFT display, I got the code ready for a pull request but I'm waiting for my other PR to get merged.

kisvegabor, 2019-07-15:
> The Wrover kit doesn't have a touch screen, it's just the TFT display,

Great, then it won't be an issue.

> I'm waiting for my other PR to get merged.

Just merged :slightly_smiling_face: 

> The TFT on the Wrover kit is 340x240, where should I update the code for this?

It should be changed in `lv_conf.h` (LV_HOR/VER_RES_MAX.

PR #13 (2019-07-18): Update README and code to support ESP_WROVER_KIT
This PR fix #11 and updates:

- The README to add information about the ESP_WROVER_KIT (tomorrow i will take a picture of it running the demo) 0d4e8fd
- Some struct initialization on `disp_spi.c` and `tp_spi.c` 3f4196d
- Fixed a problem I'm having with the paths on the `component.mk` (it was pointing one directory below the actual directory target) 18ae5c1
- Add `ILI9341_BCKL_ACTIVE_LVL` on `drv\component.mk` to let the user set the correct enable level of back light b400229
- Add `TOUCH_SUPPORT` on `drv\component.mk` to let the user enable/disa [... cut]
closed 2019-07-18 by kisvegabor