lvgl/lv_esp_idf #12: Touchscreen calibration

PKav89, 2019-07-11:
I guess this example should include a touchscreen calibration procedure. There is one in tpcal.c, I am trying to integrate it into the example, but formulas seem to be wrong. I get negative values using them.

```
- A simple equation to adjust the coordinates: x_cal = ((x_act - x1_saved) * lcd_hor_res) / (x2_saved - x1_saved);
       - x_cal: the calibrated X coordinate
       - x_act: the currently measured X coordinate
       - x1_saved, x2_saved: The raw X coordinates saved as calibration data

```
So, if I have after calibration:

`static lv_point_t point[4]; /*Calibration points: [0]: top-left; [1]: top-right, [2]: bottom-right, [3]: bottom-left */`

Which of them is x1_saved and x2_saved? And for Y?

kisvegabor, 2019-07-12:
> , I am trying to integrate it into the example, but formulas seem to be wrong. I get negative values using them.

What are the coordinates you measured and the resolution of your display?

> Which of them is x1_saved and x2_saved?

- `x1_saved` is the left coordinate
- `x2_saved` is the right coordinate
- `y1_saved` is the top coordinate
- `y2_saved` is the bottom coordinate

PKav89, 2019-07-12:
Hello kisvegabor and thanks for the best embedded graphical library in the world!

I have a 3.2" 320x240 ILI9341 LCD with XPT2046 Touch screen in landscape mode.

I get values from about 340 to about 3900 on X axis and from 200 to 3850 on Y axis. These are raw values from a touchscreen. 

The calibration routine from LittlevGL tpcal.c file collects raw coordinates of 4 points in the corners of the screen, but I cannot figure out what to do with these data. How to apply them to the raw touch screen coordinates in order to calculate true coordinates in pixels.

kisvegabor, 2019-07-15:
So in your case:
- `x1_saved = 340`
- `x2_saved = 3900`
- lets say `x_act = 1000` (should be about at ~25% in x)

```
x_cal = ((1000 - 340) * 320) / (3900-340)
x_cal = (660 * 340) / 3560
x_cal = 22400 / 3560 
x_cal = 63
```
So the pressed cooridinate is x = 63, (Really ~25% of 320)

PKav89, 2019-07-15:
Yes, that's right. I guess, it is called "linear interpolation/extrapolation". I leave my code here in case of someone will have same problem.

```
int32_t int_ext(int32_t x1, int32_t y1, int32_t x2, int32_t y2, int32_t x) 
{
	return y2 - (x2-x) / ((x2 - x1) / (y2 - y1));
}
```
And, calibration and after aquiring raw data:

```
x = int_ext(cal_data.point[0].x, CIRCLE_OFFSET + CIRCLE_SIZE / 2, cal_data.point[1].x, LCD_HOR_RES - CIRCLE_OFFSET + CIRCLE_SIZE / 2, x);
y = int_ext(cal_data.point[1].y, CIRCLE_OFFSET + CIRCLE_SIZE / 2, cal_data.point[2].y, LCD_VER_RES - CIRCLE_OFFSET + CIRCLE_SIZE / 2, y);
```

closed 2019-07-15 by PKav89

pratikmokashi, 2019-07-26:
I am having a hard time calibrating the display , The example of calibration is not working as Make fails
I tried few things but I didn't get the touch calibrated

kisvegabor, 2019-07-26:
What error message do you get exactly?

pratikmokashi, 2019-07-29:
So the demo does work, I placed a small button in the right in the middle (0,0)  of the display (320x240). when I touch on the button there is no output but if i touch the display slightly away from the button on left side it doe register. So it is issue of calibration but i have no idea how to calibrate the touch. 

#define XPT2046_AVG 4
#define XPT2046_X_MIN       1000
#define XPT2046_Y_MIN       1000
#define XPT2046_X_MAX       3200
#define XPT2046_Y_MAX       2000
#define XPT2046_X_INV       1
#define XPT2046_Y_INV       1

this is what I have in xpt2046.h

Also, I modified xpt2046.c to see the raw input from the xpt 
`if(data->state == LV_INDEV_STATE_PR)
 printf("X= %5d, Y=%5d \r\n",data->point.x, data->point.y);`

kisvegabor, 2019-07-29:
```
#define XPT2046_X_MIN 1000
#define XPT2046_Y_MIN 1000
#define XPT2046_X_MAX 3200
#define XPT2046_Y_MAX 2000
```

Are these the raw data you have measured for the corners?

pratikmokashi, 2019-07-30:
X1 = 200 , X2 = 320
Y1 = 240, Y2 = 240 

these are raw value I am getting from the function
co-ordinate in center of display (raw value) is (X=170,Y=10 to 12)

kisvegabor, 2019-07-30:
By raw value, I meant the raw value in `XPT2046`. E.g.  the x and y values [here](https://github.com/littlevgl/lv_drivers/blob/master/indev/XPT2046.c#L91). These still not scaled but really the raw data from the touch controller.

pratikmokashi, 2019-07-30:
So the raw value i got from
`xpt2046_corr(&x,&y);`
top left = (200,204)
top right =(320,196)
bottom left = (224,240)
bottom right = (320,240)

kisvegabor, 2019-07-30:
I meant BEFORE `xpt2046_corr` where the calibration values ae still not applied.

pratikmokashi, 2019-07-30:
top left = (1952,1216)
top right =(511,1184)
bottom left = (1696,503)
bottom right = (511,511)

kisvegabor, 2019-07-30:
Then you should:
```c
#define XPT2046_X_MIN 511
#define XPT2046_Y_MIN 505
#define XPT2046_X_MAX 1800 //Somewherre between 1696 and 1952
#define XPT2046_Y_MAX 1200
```

pratikmokashi, 2019-07-31:
[image]
Values in the image are from 
 ```c
xpt2046_corr(&x,&y);
```
Okay so got some output, atleast it register input on display, But there is some offset.

```c
#define XPT2046_X_MIN       511
#define XPT2046_Y_MIN       505
#define XPT2046_X_MAX       1800
#define XPT2046_Y_MAX       1500      // 1200 didnt work
```
How should I go ahead ?

ps - thank you so much for your valuable time !

kisvegabor, 2019-07-31:
It's strange. I've checked `xpt2046_corr` and seems correct. Can you print the intermediate values during the calculation?

pratikmokashi, 2019-07-31:
Yeah I can print just tell me what values you want to know, I tried multiple things but no output.
```c
static void xpt2046_corr(int16_t * x, int16_t * y)
{
#if XPT2046_XY_SWAP != 0
    int16_t swap_tmp;
    swap_tmp = *x;
    *x = *y;
    *y = swap_tmp;
#endif

    if((*x) > XPT2046_X_MIN)(*x) -= XPT2046_X_MIN;
    else(*x) = 0;

    if((*y) > XPT2046_Y_MIN)(*y) -= XPT2046_Y_MIN;
    else(*y) = 0;

    (*x) = (uint32_t)((uint32_t)(*x) * LV_HOR_RES) /
           (XPT2046_X_MAX - XPT2046_X_MIN);
           printf("cor_X= %5d",*x);   //do you want these values? x

    (*y) = (uint32_t)((uint32_t)(*y) * LV_VER_RES) /
           (XPT2046_Y_MAX - XPT2046_Y_MIN);
            printf("cor_Y= %5d",*y);  //y
#if XPT2046_X_INV != 0
    (*x) =  LV_HOR_RES - (*x);
#endif
#if XPT2046_Y_INV != 0
    (*y) =  LV_VER_RES - (*y);
#endif
}

kisvegabor, 2019-08-01:
> I tried multiple things but no output.

Probably it's because you forgot the `\n` at the end.

Let's try this:
```c
static void xpt2046_corr(int16_t * x, int16_t * y)
{
     printf("raw_X= %5d\n",*x); 
#if XPT2046_XY_SWAP != 0
    int16_t swap_tmp;
    swap_tmp = *x;
    *x = *y;
    *y = swap_tmp;
#endif
     printf("swap_X= %5d\n",*x); 

    if((*x) > XPT2046_X_MIN)(*x) -= XPT2046_X_MIN;
    else(*x) = 0;

     printf("min_sub_X= %5d\n",*x); 

    if((*y) > XPT2046_Y_MIN)(*y) -= XPT2046_Y_MIN;
    else(*y) = 0;

    (*x) = (uint32_t)((uint32_t)(*x) * LV_HOR_RES) /
           (XPT2046_X_MAX - XPT2046_X_MIN);

     printf("mult_X= %5d\n",*x); 

    (*y) = (uint32_t)((uint32_t)(*y) * LV_VER_RES) /
           (XPT2046_Y_MAX - XPT2046_Y_MIN);
#if XPT2046_X_INV != 0
    (*x) =  LV_HOR_RES - (*x);
#endif

     printf("inv_X= %5d\n",*x); 
#if XPT2046_Y_INV != 0
    (*y) =  LV_VER_RES - (*y);
#endif

     printf("\n"); 
}
```

pratikmokashi, 2019-08-01:
Alright , I will test the values and comeback.
I did add  \n later (after my reply here) got same values as before. Still I will retest once I get back to my lab.

pratikmokashi, 2019-08-02:
So I tried the following code and got these readings
```c
static void xpt2046_corr(int16_t * x, int16_t * y)
{
#if XPT2046_XY_SWAP != 0
    int16_t swap_tmp;
    swap_tmp = *x;
    *x = *y;
    *y = swap_tmp;
#endif
    if((*x) > XPT2046_X_MIN)(*x) -= XPT2046_X_MIN;
    else(*x) = 0;
    if((*y) > XPT2046_Y_MIN)(*y) -= XPT2046_Y_MIN;
    else(*y) = 0;
    (*x) = (uint32_t)((uint32_t)(*x) * LV_HOR_RES) /
           (XPT2046_X_MAX - XPT2046_X_MIN);
           printf("corrX= %d",*x);
    (*y) = (uint32_t)((uint32_t)(*y) * LV_VER_RES) /
           (XPT2046_Y_MAX - XPT2046_Y_MIN);
           printf("corrY= %5d\n",*y);
#if XPT2046_X_INV != 0
    (*x) =  LV_HOR_RES - (*x);
#endif

#if XPT2046_Y_INV != 0
    (*y) =  LV_VER_RES - (*y);
#endif
}
```
[image]

kisvegabor, 2019-08-02:
The magnitude of the numbers seems correct. From this, I can't tell more because I don't know where you touched the screen. What are the values if you touch the center of the screen?

pratikmokashi, 2019-08-02:
Its not like exact centre but it is around that area.

[image]

the values I got from the touch 

[image]

kisvegabor, 2019-08-05:
I suggest continuing it on the Forum. You can reach more people there who hopefully already had a similar issue.

Please post your last results in [Get started category](https://forum.littlevgl.com/c/get-started).