lvgl/lv_esp_idf #16: Touch works only with the tpcal example

winny2809, 2019-09-19:
I imported this library in a platform underesp-idf environment.
All the graphic functions on the display work correctly, but unfortunately the touch in none of the examples works.
Only by calling the "tpcal_create ()" function for touch adjustment, the device receives the signal that the screen has been pressed. It is as if only the "touch" event is called in this function. How is it possible that only in this example does the touch work correctly?

Thanks, and congratulations for the great work you have done.

embeddedt, 2019-09-19:
@winny2809 It's likely that your touch screen is not calibrated correctly, so you never touch any of the objects on the screen (e.g. buttons). However, the `tpcal` example is designed to calibrate the screen so it will register touches anywhere.

winny2809, 2019-09-19:
The problem is that even creating an object with the call function, when you touch any point on the screen no event is generated, an event is generated only by using the "tpcal_create" function.
It is as if only the "tpcal_create" function read the interrupt from the touch screen.
Why?

kisvegabor, 2019-09-19:
`tp_cal` checks to the whole display area for clicks. I suggest to `printf` the measured raw touchpad data fro your driver to if it's correct.

embeddedt, 2019-09-19:
> The problem is that even creating an object with the call function, when you touch any point on the screen no event is generated, an event is generated only by using the "tpcal_create" function.

That is because, as I said above, in the other examples there are small objects on the screen in various places. It just so happens that the current data you are reading from your touchscreen is not hitting those areas.

However, think of `tpcal` as creating a huge button that covers the whole screen. No matter where the data you read claims you touch, the huge button will always get clicked as it fills the whole touchscreen.

As @kisvegabor said it's likely that the coordinate data you are reading is bogus.

winny2809, 2019-09-19:
And then? in this case how should I proceed? Excuse me

embeddedt, 2019-09-19:
> in this case how should I proceed?

First, let's figure out whether it's bogus or not. That is, does touching the top left of the screen give you roughly `(0,0)`, and does touching the bottom right give you a coordinate around the screen width and height (i.e. for 240x320 `(240, 320)`)?

winny2809, 2019-09-19:
[image]

This is the result

kisvegabor, 2019-09-19:
It's cleary the issue of the driver. Probably you read always 0xFF. Typically it indicates a problem with the SPI. (wrong pins or wrong configuration)

winny2809, 2019-09-19:
I have the doit devkit v1 card.
My configuration is:

#define XPT2046_IRQ 25

#define ENABLE_TOUCH_INPUT  1

#define TP_SPI_MOSI 23
#define TP_SPI_MISO 19
#define TP_SPI_CLK  18
#define TP_SPI_CS   5

winny2809, 2019-09-19:
the vspi configuration is:

void tp_spi_init(void)
{

	esp_err_t ret;

	spi_bus_config_t buscfg={
		.miso_io_num=TP_SPI_MISO,
		.mosi_io_num=TP_SPI_MOSI,
		.sclk_io_num=TP_SPI_CLK,
		.quadwp_io_num=-1,
		.quadhd_io_num=-1
	};

	spi_device_interface_config_t devcfg={
		.clock_speed_hz=10*1000*1000,           //Clock out at 80 MHz
		.mode=0,                                //SPI mode 0
		.spics_io_num=-1,              //CS pin
		.queue_size=1,
		.pre_cb=NULL,
		.post_cb=NULL,
	};

	//Initialize the SPI bus
	ret=spi_bus_initialize(VSPI_HOST, &buscfg, 2);
	assert(ret==ESP_OK);

	//Attach the LCD to the SPI bus
	ret=spi_bus_add_device(VSPI_HOST, &devcfg, &spi);
	assert(ret==ESP_OK);
}

kisvegabor, 2019-09-19:
Let's continue in the [Forum](https://forum.littlevgl.com) because more users read it.

closed 2019-09-19 by kisvegabor