lvgl/lv_esp_idf #17: Error when compiling

Terixxx, 2019-09-26:
Hello,
i am facing following error when try to build the project: 
_esp-idf/esp32/libesp32.a(cpu_start.c.obj):(.literal.main_task+0x10): undefined reference to `app_main'
esp-idf/esp32/libesp32.a(cpu_start.c.obj): In function `main_task':
C:/Users/asenn/Desktop/esp-idf-2/components/esp32/cpu_start.c:514: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status_

I use this guide as in the readme file:
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html
but can't 'figured out what is wrong.

embeddedt, 2019-09-26:
I don't use ESP32, but I frequently see that people have issues if they use a version that is too new. Maybe you should try an older version of ESP-IDF.

Terixxx, 2019-09-26:
I am wondering this project with which IDE is created, which compiler is used and the OS?

kisvegabor, 2019-09-26:
> I am wondering this project with which IDE is created, which compiler is used and the OS?

It's a simple makefile project but as @embeddedt mentioned ESP-IDF's build has completely changed. Therefore the project is not working with the latest version of ESP-IDF.

Take a look at this post in the Forum:  
https://forum.littlevgl.com/t/how-to-create-complex-project-structures-in-esp-idf/619/15?u=kisvegabor

I'm using v3.2 of ESP-IDF which works well with this project.

Terixxx, 2019-09-27:
I tried with v3.2, but still can't compile. The error is the same.

kisvegabor, 2019-09-27:
Please open a topic in the Forum about this question. 
The Forum is visited by more people, so there is a higher chance to get help there.
https://forum.littlevgl.com/

closed 2019-09-27 by kisvegabor