lvgl/lv_esp_idf #20: KConfig?

jeremyjh, 2019-10-02:
The standard convention in ESP-IDF libraries is to use KConfig / menuconfig / sdkconfig for user configuration of libraries/components. Right now this project uses a couple of makefile flags as well as header defines for configuration of the library for the end-user's device. I would suggest a move to use all header defines but have those defines in turn reference sdkconfig variables. 

So in the header for example we'd change this:

```c
#define DISP_SPI_MOSI 23
```

to 

```c
#define DISP_SPI_MOSI CONFIG_DISP_SPI_MOSI
```

`CONFIG_DISP_SPI_MOSI` in turn is set in sdkconfig directly or through menuconfig, and will still have a default value. The two layers of defines gives you the flexibility to quickly change it in the library header while testing, which does not recompile the whole sdk.

The advantage for the user is this similar to other libraries as well as all the IDF components themselves, but also that they do not necessarily have to vendor the library, they can keep it as a git submodule without worrying about the dirty flags.

I very recently [implemented such a scheme](https://github.com/jeremyjh/ESP32_TFT_library/blob/master/components/tft/Kconfig) for the loboris driver and would be happy to submit a PR for this library as well.

kisvegabor, 2019-10-02:
> I very recently implemented such a scheme for the loboris driver and would be happy to submit a PR for this library as well.

It sounds very well. Please send a PR!

jeremyjh, 2019-10-23:
I'm pretty close to having a PR ready, should be in the next couple days, all that is left are instructions in the README. WIP is on this branch: https://github.com/jeremyjh/esp32_ili9341/tree/kconfig

kisvegabor, 2019-10-24:
Awesome, thank you for the feedback.

Do you know about the new Cmake based build system of ESP? If so have you already make it work this project?

jeremyjh, 2019-10-24:
Yes Cmake support is in that branch too.

kisvegabor, 2019-10-24:
Cool :sunglasses:

PR #25 (2019-10-31): Kconfig
resolves #20 

Implements suggestions from #20 to use KConfig for end-user configuration. It also includes support and instructions for CMake (when using IDF 4.x). Part of the reason for wanting the KConfig changes is so that this repository can be used as a library submodule without copying or changing any files and thus keeping the submodule status clean. This makes updates easier and encourages reproducible builds. There are other minor changes to build files and paths to support this goal. A good way to get an overview is to look at the new instructions in the [rendered README](htt [... cut]
closed 2019-10-31 by kisvegabor