lvgl/lv_esp_idf #22: line width for style not always functioning correctly

fivdi, 2019-10-20:
Below is a variation of [the main program from this repository](https://github.com/littlevgl/esp32_ili9341/blob/master/main/main.c). The goal of the program is to display something similar to a big wide letter "W" at the top of the display.

If the line width (`style_letter_w.line.width`) for the lines making up this big wide letter "W" is 1, 2, 3 or 4, the program functions correctly. If the line width is 5 the "W" is no longer displayed correctly. What appears on the display is not even close to what it should be.

The tests were performed on an ESP-WROVER-KIT v4.

```c
/* SPI Master example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "driver/gpio.h"
#include "lvgl/lvgl.h"
#include "lv_examples/lv_apps/demo/demo.h"
#include "esp_freertos_hooks.h"

#include "drv/disp_spi.h"
#include "drv/ili9341.h"
#include "drv/tp_spi.h"
#include "drv/xpt2046.h"

static void IRAM_ATTR lv_tick_task(void);

static lv_obj_t *letter_w;

void ui_create()
{
  // The points for a big wide letter "w" that will be displayed at the top
  // of the display.
  static lv_point_t letter_w_points[] = {
    {5, 5}, {70, 70}, {120, 10}, {180, 60}, {240, 10}
  };

  static lv_style_t style_letter_w;
  lv_style_copy(&style_letter_w, &lv_style_plain);
  style_letter_w.line.width = 5; // ***************************************************
                     // If style_letter_w.line.width is set to 5 the program breaks
                     // If style_letter_w.line.width is set to 3 the program functions

  letter_w = lv_line_create(lv_disp_get_scr_act(NULL), NULL);
  lv_line_set_points(letter_w, letter_w_points, 5);
  lv_line_set_style(letter_w, LV_LINE_STYLE_MAIN, &style_letter_w);
  lv_obj_align(letter_w, NULL, LV_ALIGN_IN_TOP_MID, 0, 20);
}

void app_main()
{
  lv_init();

  disp_spi_init();
  ili9341_init();

#if ENABLE_TOUCH_INPUT
  tp_spi_init();
  xpt2046_init();
#endif

  static lv_color_t buf1[DISP_BUF_SIZE];
  static lv_color_t buf2[DISP_BUF_SIZE];
  static lv_disp_buf_t disp_buf;
  lv_disp_buf_init(&disp_buf, buf1, buf2, DISP_BUF_SIZE);

  lv_disp_drv_t disp_drv;
  lv_disp_drv_init(&disp_drv);
  disp_drv.flush_cb = ili9341_flush;
  disp_drv.buffer = &disp_buf;
  lv_disp_drv_register(&disp_drv);

  // Set TOUCH_SUPPORT on drv\component.mk to 1 if
  // your board have touch support
#if ENABLE_TOUCH_INPUT
  lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.read_cb = xpt2046_read;
  indev_drv.type = LV_INDEV_TYPE_POINTER;
  lv_indev_drv_register(&indev_drv);
#endif

  esp_register_freertos_tick_hook(lv_tick_task);

  ui_create();

  while(1) {
    vTaskDelay(1);
    lv_task_handler();
  }
}

static void IRAM_ATTR lv_tick_task(void)
{
  lv_tick_inc(portTICK_RATE_MS);
}```

kisvegabor, 2019-10-21:
It should be already fixed. Please update `lvgl` to the latest `master`.

fivdi, 2019-10-21:
Yes, updating `lvgl` to the latest `master` resolves the issue. Thank you very much. I guess the fix will be available in the next official release of lvgl so I'll go ahead and close the issue.

closed 2019-10-21 by fivdi

kisvegabor, 2019-10-21:
> I guess the fix will be available in the next official release of lvgl so I'll go ahead and close the issue.

Sure!

embeddedt, 2019-10-21:
> Yes, updating `lvgl` to the latest `master` resolves the issue.

@fivdi As a note, it's usually wise whenever downloading one of our example projects to run a `git pull` on the submodules, just to make sure they're up to date.

fivdi, 2019-10-21:
@embeddedt Thanks for the tip.