lvgl/lv_esp_idf #24: Problems in SPI driver not setting DC correctly

Boldie, 2019-10-24:
Just for information, I used your driver to bring my stuff working. Since I rewrote big parts of them (I use the same SPI for touch and display), I found some problems I want to let you know.

In the display driver, there is a problematic condition on fast systems switching the DC pin, before the SPI transfer has been finished, especially in the beginning of the transfer. Since SPI returns after sending the stuff to the spi_device_queue, the pin might become toggled before the SPI transfer has been finished. Before triggering the spi ttuff inside `ili9341_send_*` a `while(spi_trans_in_progress); `is needed.

kisvegabor, 2019-10-25:
Can you send a pull request to fix it?

Boldie, 2019-10-25:
Ohh, I rewrote big parts of the driver part to support xpt2046 calibration, storing calibration to NVS, using a single spi bus (especially this is incompatible with yours, but can be changed to support both variants easily), KConfig, make gpio-pin configuration also working for JTAG pins, removed stack usage for spi_device_queue_trans (may create a crash), ... 
But maybe it is of interest for others, I uploaded it as separate repo: https://github.com/Boldie/lv_drv

kisvegabor, 2019-10-28:
I've examined this issue. As I see it can cause a problem only in `ili9341_flush`. However, LittlevGL won't call a new `ili9341_flush` while `lv_disp_flush_ready()` is not called in `spi_ready`.  So theoretically it really would be wrong (as you mentioned), but int the practice it should work fine.

Do you think it's still should be still fixed?

Boldie, 2019-10-28:
No, this is not correct, because for an update you have to trigger more transfers for one flush:

```
	/*Column addresses*/
	ili9341_send_cmd(0x2A); <- GPIO DC needs to be 0
...
	ili9341_send_data(data, 4);  <- GPIO DC needs to be 1

	/*Page addresses*/
	ili9341_send_cmd(0x2B);   <- GPIO DC needs to be 1
...
	ili9341_send_data(data, 4);

	/*Memory write*/
	ili9341_send_cmd(0x2C);
```

And in my case, the DC was reset to early and the next command was not understood as it should be.
```
	gpio_set_level(CONFIG_LVGL_DRV_ILI9341_DC, 0);	 /*Command mode*/
	disp_spi_send_data(&cmd, 1); <<< This line returns immediately, but the transfer is not done. The DC can now be reset by the next ili9341_send_data and thus the command is not being recognized as command, it is recognized as data.
```

I have added a LA to the ESP and I saw the above described case. DC was set to 1 while the command is being sent out by the SPI interface.

kisvegabor, 2019-10-28:
You are right. I'll update it.

kisvegabor, 2019-10-29:
I've fixed it! Thank you for pointing out this issue.

commit a0c6db7f (2019-10-29): spi: wait for transfer ready before changinf DC. Fixes #24
closed 2019-10-29 by kisvegabor