lvgl/lv_esp_idf #28: Add ranges to Kconfig integer parameters?

C47D, 2019-11-04:
Hi @jeremyjh, should we add ranges to the display pin configurations? So the users doesn't configure invalid pins?

In espressif examples they use them like so:
```
menu "Example Configuration"

    config BLINK_GPIO
        int "Blink GPIO number"
        range 0 34
        default 5
        help
            GPIO number (IOxx) to blink on and off.

            Some GPIOs are used for other purposes (flash connections, etc.) and cannot be used to blink.

            GPIOs 35-39 are input-only so cannot be used as outputs.

endmenu
```

jeremyjh, 2019-11-05:
Yes makes sense to me.

C47D, 2019-11-05:
Will try to add them in a PR later this week.

Do you have any other ideas to improve the repo? I was thinking on adding and option on the Kconfig to choose which example to build, but I don't know how easy can it be done.

jeremyjh, 2019-11-05:
One thing the library still needs is rotation support. I haven't needed it myself yet though so I haven't implemented.

I think the demo idea is a good one. You could do it with a menu choice similar to how the predefined display works. If you do that it that the Kconfig for it should go into the main/ folder as it isn't applicable to the library, only to the demo project. I'd call the menu something like LittleVgl demo. Then the main.c needs to have #ifdefs on those variables that determine which demo function to run.

C47D, 2019-11-05:
Do you happen to know if there is anything on the Kconfig with exclusive options? So only one demo project can be selected and not multiple.

jeremyjh, 2019-11-05:
Thats how a choice group works, if you follow the example of the predefined display it works that way too.

C47D, 2019-11-08:
Closed with #29

closed 2019-11-08 by C47D