lvgl/lv_esp_idf #33: [META] Add issue template?

C47D, 2019-11-12:
Hi,

I would like to add an issue template to help users give us valuable information when they can't make the project run on their setup.

What should we ask for? Here's what i have so far...

- Symptoms (display flickering, display all black/white)?
- ESP32 Chip.
- ESP-IDF version.
- Development kit.
- Development machine OS.
- Compilation warnings/errors if available.
- File with the compilation log?

On the README is a paragraph about using this repo as a way to add lvgl to custom user projects, in that case I think we should ask for the user directory structure.

Any aditional ideas?

jeremyjh, 2019-11-12:
I prefer the "Expected Results", and "Actual Results" versus Symptoms terminology. It can cut down on a lot of misunderstandings. But I think this looks good.

C47D, 2019-11-12:
Agree, I will send a PR to close this.

C47D, 2019-11-13:
Fixed with #34

closed 2019-11-13 by C47D