lvgl/lv_esp_idf #35: Settings via 'menuconfig' for touch panel use are ignored

patrickdittrich, 2019-11-19:
Changes from #32 do conditionally include/exclude header files used for touch panel use depending on ENABLE_TOUCH_INPUT. 
But ENABLE_TOUCH_INPUT is only defied in 'tp_spi.h' with the value from CONFIG_LVGL_ENABLE_TOUCH (set via 'menuconfig'), so it will never be included. 
See line 24 in 'main.c' 15a15d01d9b287afa43df5a6c515dcd45c4795cc

jeremyjh, 2019-11-19:
Good catch. So basically right now the demo won't ever enable the touch controller. @C47D did we actually need to only conditionally include those headers? Was there a warning that this resolved?

C47D, 2019-11-19:
Hi,

I actually used it to avoid registering `lv_indev_drv_t indev_drv` when using the ESP32-WroverKit v4.1 (which soesn't include the touch controller).
https://github.com/littlevgl/lv_port_esp32_ili9341/blob/d797cb40dcf57d357fd530c9e8a40fd5872b3bd9/main/main.c#L54-L60

Should we apply the following patch?
```diff
diff --git a/main/main.c b/main/main.c
index 35d47ef..cabb2ec 100644                                                                                        --- a/main/main.c
+++ b/main/main.c
@@ -21,10 +21,8 @@
 #include "disp_spi.h"
 #include "ili9341.h"

-#if ENABLE_TOUCH_INPUT
 #include "tp_spi.h"
 #include "xpt2046.h"                                                                                                -#endif

 static void IRAM_ATTR lv_tick_task(void);

@@ -51,7 +49,7 @@ void app_main()
        disp_drv.buffer = &disp_buf;
        lv_disp_drv_register(&disp_drv);

-#if ENABLE_TOUCH_INPUT
+#if (ENABLE_TOUCH_INPUT == 1)
     lv_indev_drv_t indev_drv;
     lv_indev_drv_init(&indev_drv);
     indev_drv.read_cb = xpt2046_read;
```

jeremyjh, 2019-11-19:
The #ifdef related to the driver registration in line 53 was already there, the [patch in question](https://github.com/littlevgl/lv_port_esp32_ili9341/commit/15a15d01d9b287afa43df5a6c515dcd45c4795cc) only made the #includes conditional, but they are conditional on a macro set in that same include file, so basically it will always be defined and the touch panel will never be registered. I don't think we need the includes to be conditional.

C47D, 2019-11-19:
If we compare the value of ENABLE_TOUCH_INPUT to 0 or 1 the code sorrounded by the `#if` will only be active if the user enables the touch input, otherwise (the `#if` alone) that code will always be enabled, with touch input enabled or not. Please let me know if I'm wrong.

jeremyjh, 2019-11-19:
#if will include the block if the value of the [expression is non-zero](https://gcc.gnu.org/onlinedocs/cpp/If.html). If ENABLE_TOUCH_INPUT is 0, the code is not included, otherwise it is. I added this code in an earlier PR because on the M5 if you enable the touchpad driver with default settings there is a GPIO conflict and you'll never even see anything appear on the screen.

C47D, 2019-11-19:
Sorry, didn't knew that, just submitted a PR trying to solve the issue, please let me know if it does.

PR #36 (2019-11-20): Fix touch ignored
Fixes #35
commit 3633ccd2 (2019-11-20): Fixes #35, touch input ignored
closed 2019-11-20 by kisvegabor