lvgl/lv_esp_idf #37: How to ROTATE the screen

donny681, 2019-12-04:
I want to ratate the screen to 90,How to do it ?

jeremyjh, 2019-12-04:
This is not implemented in the driver yet. You can use the driver functions to send a rotation command though - it is a 0x36 command and we use an example here: https://github.com/littlevgl/lv_port_esp32_ili9341/blob/master/components/lvgl_ili9341/ili9341.c#L108

To determine the correct data command to send, you can refer to this code. https://gist.github.com/motters/38a26a66020f674b6389063932048e4c#file-ili9844_rotation-h-L4

donny681, 2019-12-05:
I try to determine the data of 0x36 command.But it may be useless.

donny681, 2019-12-12:
I try this,it is ok.Thanks

embeddedt, 2019-12-12:
Great!

I'll close this issue then.

closed 2019-12-12 by embeddedt