lvgl/lv_esp_idf #39: Program crashes when adding wifi support

cbarransunny, 2019-12-07:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**
added wifi station code from esp-idf mqtt example:
https://github.com/espressif/esp-idf/blob/release/v3.3/examples/protocols/mqtt/tcp/main/app_main.c#L171

**Code to reproduce the issue**
  esp_register_freertos_tick_hook(lv_tick_task);
  xTaskCreate(display_task, "display_task", 2048, NULL, 5, NULL); 
  esp_err_t error_test = nvs_flash_init();
  if(error_test == ESP_OK)
  {
    wifi_init_sta();
  }
  while(1)
  {
		vTaskDelay(1000/portTICK_PERIOD_MS);
	}
}

static void display_task(void *pvParameter)
{
  Demo();
  while(1) {
		vTaskDelay(5/portTICK_PERIOD_MS);
		lv_task_handler();
	}

}
**Expected Results**
The display shows to demo and wifi connects
**Actual Results**
Guru Meditation Error: Core 0 panic'ed (Cache disabled but cached memory region accessed)
**ESP32 Chip version**

**ESP-IDF version**
3.3
**Development kit used**
NodeMCU 32s
**Development machine OS**
Windows 10, Platoformio
**Compilation warnings/errors (if available)**

**If possible, copy the compilation log into a file and attach it here**

embeddedt, 2019-12-28:
Not stale. (I can't help with this issue as I don't use ESP32.)

embeddedt, 2020-01-16:
@kisvegabor @amirgon Perhaps one of you can help with this?

amirgon, 2020-01-16:
Where is the stack trace?
On a crash report, the very minimum is providing a parsed stack trace.

justbarran, 2020-01-17:
Problem was solved by running Littlevgl only on core 1 while wifi and everything else runs on core 0. And using esp_timer_start_periodic(_lv_tick_timer, LV_TICK_MS*1000) to run the ticker.

embeddedt, 2020-01-17:
It's probably a concurrency issue (LittlevGL needs you to [wrap calls with some type of synchronization primitive](https://docs.littlevgl.com/en/html/porting/os.html) to work with multithreaded systems).

jeremyjh, 2020-01-17:
I should have noticed that just looking at this code, I had a similar thing happen to me. I use a pinned task to run the lv_task_handler(). I felt better about that approach than adding a mutex because there is absolutely no chance of a deadlock.

justbarran, 2020-01-18:
Yea the main problem came from esp_register_freertos_tick_hook(lv_tick_task); This should never be used. And I used a Recursive mutex around the LittleVGL functions.

monkeytronics, 2020-01-23:
Hey guys. I've just got this up and running on ESP32, but am crashing every couple of minutes... I am using esp_register_freertos_tick_hook(lv_tick_task); So, I guess I'm not doing it right?! I posted the stack trace and some extra info here:

https://github.com/espressif/esp-iot-solution/issues/37#issuecomment-577487967

Any chance you guys know what I'm doing wrong?? Thanks all.

amirgon, 2020-01-23:
> Any chance you guys know what I'm doing wrong?? Thanks all.

- One thing I don't understand about your code: from the stack trace it looks like `ili9341_flush` is calling `vPortYield`.
However, I don't see where `vPortYield` is called anywhere in this project.  
Did you change ili9341? Where are your sources?

- Try to run it all on a single core (disable the second core when configuring ESP32) and check if the problem still happens. Even if the problem is still there, it would be easier to debug.

monkeytronics, 2020-01-23:
Hmmm. I just downloaded the ili9341 code from here. I haven't touched it at all.

I've just built it again on a single core like you suggested. It's been running for a half hour now without crashing...  It looks encouraging that you've fixed the immediate fault. Any idea why it might have been crashing with the dual core. Suggestions above that using esp_register_freertos_tick_hook(lv_tick_task) is a bad thing. Any clue as to why that is? Is there a better ESP32 approach somewhere? I don't really need the processing power as it happens, but keen to keep my code robust if possible... 

Cheers,

amirgon, 2020-01-23:
> Hmmm. I just downloaded the ili9341 code from here. I haven't touched it at all.

Your stack trace:
> 0x4009742d: vPortYield at C:/msys32/home/thebi/esp/esp-idf/components/freertos/portasm.S:555
> 0x40142d70: ili9341_flush at C:/msys32/home/thebi/esp/esp-jumpstart/components/lvgl_ili9341/**ili9341.c:130**

Code from here:
https://github.com/littlevgl/lv_port_esp32_ili9341/blob/edf5bf19083e911876318c0842c091a6dfdc6d24/components/lvgl_ili9341/ili9341.c#L127-L137

**That doesn't add up**. line 130 on `ili9341.c` is an empty line, but on your stack trace it's calling `vPortYield`.  
Are you aligned to HEAD?

> Suggestions above that using esp_register_freertos_tick_hook(lv_tick_task) is a bad thing

I would try using `xTimerCreate` for calling `lv_tick_inc`, and `xTaskCreate` pinned to core for calling `lv_task_handler` (with some delay inside to yield).

I'm doing something like that on [modlvesp32](https://github.com/littlevgl/lv_binding_micropython/blob/bf3f241327fe5089f1f9af234e98707a3aa33d1e/driver/esp32/modlvesp32.c#L31-L64) (although I'm using micropython scheduler instead of `xTaskCreate`, but under the hoods micropython is calling `xTaskCreate`).

closed 2020-02-20 by stale[bot]