lvgl/lv_esp_idf #40: leak memony

donny681, 2019-12-12:
I try to decode several png images.However,when I decode over two images,it will reduce the heap size gradully.It is ok when I decode only one image.Do you know the reason?

source code:

` 
 CURRENT_img1 = lv_img_create(lv_scr_act(), NULL);
   
 lv_obj_set_pos(CURRENT_img1, 10, 65);

   lv_img_set_src(CURRENT_img1, "/spiffs/wet/100.png");

 lv_obj_t * NEXT_day_weather_img = lv_img_create(lv_scr_act(), NULL);

    lv_obj_set_pos(NEXT_day_weather_img, 0, 185);

    lv_img_set_src(NEXT_day_weather_img, "/spiffs/wet/101.png");
` 
 print the heap size every second.
 
debug:

`app_main buffer get: 4170120

app_main buffer get: 4168480

app_main buffer get: 4165284

app_main buffer get: 4162092

app_main buffer get: 4158896

app_main buffer get: 4155700

app_main buffer get: 4152504

app_main buffer get: 4147752

app_main buffer get: 4146112

app_main buffer get: 4142916

app_main buffer get: 4139720

app_main buffer get: 4136524

`

donny681, 2019-12-12:
When memory is exhausted,it output this debug:

Warn: Image draw cannot open the image resource 	(/home/fyy/esp/project/lv_port_esp32_ili9341/components/lvgl/lvgl/src/lv_draw/lv_img_cache.c #118)
Warn: Image draw error 	(/home/fyy/esp/project/lv_port_esp32_ili9341/components/lvgl/lvgl/src/lv_draw/lv_draw_img.c #61)
error 83: memory allocation failed

donny681, 2019-12-12:
I found that if I create over two images objects,it will recall decoder_open function all the time,which reduce the memories.

embeddedt, 2019-12-12:
I can't see enough of your code to know exactly what you're doing, but it sounds like you are creating new image objects over and over. You don't need to do that. Just change the source of the existing image object(s).

Again, not much code to work with here, but I'm guessing that your UI has current and next weather images? That means you only need two image objects, and you can just change the sources of them as necessary. You don't need to create new ones each time you want a different image.

If, for some reason, you really need to create a new image object, remember that you should delete the first one with `lv_obj_del`. Otherwise it stays on the screen and keeps consuming memory.

donny681, 2019-12-13:
Actually,I do need to create several images and change them day by day. Like you say,If I delete the images,how can I change the source of pictures.

donny681, 2019-12-13:
I find the real reason,when I create two imge objects and a scrolled label ,there is a conflict between image and scrolled label,I post the code here (https://github.com/donny681/littlevgl_esp32).You can see that the it will reduce memories  gradually.

embeddedt, 2019-12-13:
> I do need to create several images and change them day by day.

As I mentioned, you can change the source of an image after it's created by calling `lv_img_set_src` again. There is **no need** to recreate/delete the image object.

Hmm. I don't see anywhere in your code that you are changing the image. You seem to just create it once in `my_demo_create` and then do nothing afterwards. Is `my_demo_create` ever called again after [this line](https://github.com/donny681/littlevgl_esp32/blob/master/main/main.c#L142)?

> You can see that the it will reduce memories gradually if it creates two image object and a scrolled label.

I very much doubt this is the cause. It's more likely that the label just speeds up the exhaustion of heap space because it uses some memory to store the text.

Please check that you are not recreating objects multiple times. LittlevGL does not work like some other libraries where you have to call a "drawText" API or something similar in a loop. For LittlevGL creating an object *once* and then updating it later

donny681, 2019-12-16:
No,I think that you misunderstand my meaning.As you mentioned.I do nothing after creating the UI.However,I don't know the reason why it reduces memories  every second If I create two images and a scrolled label.There are five steps to prove the problems.

Firstly,just create two images then do nothing,the memonies are stable for a long time after a few seconds.

Secondly,If I create two images and a scrolled label .However,it resuces the memonies every second.After a few minutes all the memonies are exhausted because the scrolled label consumes the memonies every second. 

Thirdly,if I just create only one images and a scrolled label,the memonies are still  stable for a long time.

Forthly,if I  just create a scrolled label,the memonies are still  stable for a long time after a few seconds.

Finally,If I create several images and the memonies are still  stable for a long time after a few seconds.

kisvegabor, 2019-12-16:
@donny681 

I was able to reproduce the issue and found the problem. `lodepng` loads the PNG file into buffer, and this buffer wasn't freed.

I've fixed it here: https://github.com/littlevgl/lv_lib_lodepng/commit/5018378ee704be467866609a259dc9b1161482a5

I hope it solves your issue.

donny681, 2019-12-17:
@kisvegabor it works.Thank you.

donny681, 2019-12-17:
@kisvegabor However,when I try to change the text of other static  label.It will affect the original scrolled label .For example,if I create two images ,a scrolled label and a static label.If I want to change the text of the static label.Actually it change the text of scrolled label.

kisvegabor, 2019-12-17:
Please send a code snippet?
Do you see the issue without the images too?

donny681, 2019-12-24:
Sorry,It is my fault .I do something wrong with it personally.

closed 2019-12-24 by donny681