lvgl/lv_esp_idf #41: Converted font won't compile as-is

rbarris, 2019-12-14:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**
I used the LittlevGL online font converter tool to convert a font.
I downloaded the resulting C file and added it to my project incorporating the lv_port_esp32_ili9341.
The font source isn't fully compatible with this version of LittlevGL and had errors I had to fix.
It is not actually clear to me whether this is an issue in the font converter being too new, or the code in this port being different, not sure.  The errors are minor, here is the diff

rbarris@oryx-i7:~/sandbox/ray/esp1$ diff ~/Downloads/build.c ~/sandbox/ray/esp1/main/build.c
1377c1377
<     .class_pair_values   = kern_class_values,
---
>     .class_pair_values   = (const uint8_t *)kern_class_values,
1412c1412
<     .subpx = LV_FONT_SUBPX_NONE,
---
> //    .subpx = LV_FONT_SUBPX_NONE,

**Code to reproduce the issue**
[font edits.zip](https://github.com/littlevgl/lv_port_esp32_ili9341/files/3964227/font.edits.zip)

**Expected Results**
A converted font should integrate smoothly with LVGL

**Actual Results**
minor compile errors

**ESP32 Chip version**
m5stack gray

**ESP-IDF version**
ESP-IDF v4.1-dev-1086-g93a8603c5-dirty

**Development machine OS**
Ubuntu 17.10

embeddedt, 2019-12-14:
This is because the font converter has been updated to support LittlevGL 6.1, but there are a few minor incompatibilities with 6.0.x now. There are two possible solutions:

* Fix the minor incompatibility the way you did, and continue to use LittlevGL 6.0.
* Upgrade to LittlevGL 6.1; it's a minor release so you shouldn't need to do much work to upgrade.

kisvegabor, 2019-12-16:
As the question is answered I close this issue.

closed 2019-12-16 by kisvegabor