lvgl/lv_esp_idf #42: Support IL9488

rdelcorro, 2019-12-18:
Is it possible to support the IL9488? As far as I understand, the differences are quite minimal. Thanks

kisvegabor, 2019-12-19:
It's not supported by default, but it'd be great if you could add it. 

It can be a [Kconfig](https://github.com/littlevgl/lv_port_esp32_ili9341/blob/master/components/lvgl_ili9341/Kconfig) option the choose between the drivers.

cc
@jeremyjh @C47D

C47D, 2019-12-19:
Hi,

I don't have any ILI9488 at hand but I will check the datasheet for it, so if @rdelcorro have one it would be very helpful if you can help us with the validation.

In regards of choosing the LCD driver on the esp configuration it's possible, but I would like to ask @jeremyjh for suggestions on how to implement it, I would suggest to abstract the LCD driver, something like this:

```diff
diff --git a/main.c b/main.c                                                                                                          index 6fc45cb..2d98823 100644                                                                                                         --- a/main.c                                                                                                                          +++ b/main.c
@@ -11,7 +11,7 @@

 #include "disp_spi.h"
-#include "ili9341.h"
+#include "disp_driver.h"

 #include "tp_spi.h"
 #include "xpt2046.h"
@@ -23,7 +23,7 @@ void app_main()
        lv_init();

        disp_spi_init();
-       ili9341_init();
+       disp_driver_init();

 #if ENABLE_TOUCH_INPUT
        tp_spi_init();
@@ -37,7 +37,7 @@ void app_main()

        lv_disp_drv_t disp_drv;
        lv_disp_drv_init(&disp_drv);
-       disp_drv.flush_cb = ili9341_flush;
+       disp_drv.flush_cb = disp_driver_flush;
        disp_drv.buffer = &disp_buf;
        lv_disp_drv_register(&disp_drv);
```

I have some free time this saturday, I can work on this then.

Regards

rdelcorro, 2019-12-19:
Thanks @C47D, I can help with the validation.

C47D, 2019-12-23:
> Is it possible to support the IL9488? As far as I understand, the differences are quite minimal. Thanks

Hi, do you happen to know the differences between the ILI9341 and ILI9488 drivers? I'm still stuck trying to figure out how to setup the project configuration to be able to choose between diferent drivers.

rdelcorro, 2019-12-23:
@C47D this is the aliexpress link: https://www.aliexpress.com/item/33057812377.html?spm=a2g0s.12269583.0.0.54b636d3APjqbt

I was also able to use the screen with this code: https://gist.github.com/rdelcorro/beb7ebe929064797b926f3affcd9e0ac which uses this library: https://github.com/Bodmer/TFT_eSPI. 

As far as I understand, the difference is: 
- Initialization sequence (can be seen in the repo above) 
- 16 bit color mode (my screen) but it supports up to 24

Thanks!

C47D, 2019-12-23:
@rdelcorro Thanks for the information, I'm still trying to figure out how to handle the driver abstraction and also learn how to use the Kconfig lang to extend the menuconfig interface. Please give me some more time to arrive to something that handle your display driver and maybe others in the future.

jeremyjh, 2019-12-23:
@C47D I think I'd just define some constants for the driver such that ILI9341 = 1, ILI9488 = 2 and then use a menu choice group to select from the two options, similar to how the pre-defined displays work. In the driver code we could have multiple init functions defined and call the correct one based on this value.  

My fork of the loboris TFT driver has a KConfig for exactly that purpose: https://github.com/jeremyjh/ESP32_TFT_library/blob/master/components/tft/Kconfig#L31

We use the config value to define a controller (defaults to ILI9341): https://github.com/jeremyjh/ESP32_TFT_library/blob/master/components/tft/tftspi.h#L210

Then we use that value to select the init commands: 
https://github.com/jeremyjh/ESP32_TFT_library/blob/master/components/tft/tftspi.c#L908

C47D, 2019-12-24:
Thanks for the links @jeremyjh, it's much clearer now.

I see you share the SPI and control signals across the drivers:

https://github.com/jeremyjh/ESP32_TFT_library/blob/cf347104a635a876fe9614ff465d5b9dec10f255/components/tft/tftspi.h#L54-L64

After checking the ILI9488 datasheet and @rdelcorro ali link we can notice it in fact uses the same pins as the ILI9341 driver, so i think the first step is to modify below pin naming so it becomes more generic? Or should i add an .h and .c files for the ILI9488?

https://github.com/littlevgl/lv_port_esp32_ili9341/blob/1f2a6f3790244cd39c58d02b246580be6b45c9b1/components/lvgl_ili9341/ili9341.h#L24-L26

C47D, 2020-01-16:
We're still trying to make this work...

Mat-Alm, 2020-01-23:
Hi! I'am trying to port this display (ILI9488) to littlevgl. Well I made the initialization sequence works fine.
But now the problem is this: 

ILI9488 only support RGB666 (18bits) it's like RGB888 but with some bits unused... 

How to add support for littlevgl RGB sapce conversion?

Mat-Alm, 2020-01-23:
Hi! Managed to get it working! Bellow are my changes...

In ILI9341.h change 

DISP_BUF_SIZE (LV_HOR_RES_MAX * 20)

In ILI9341.h add

```
/*******************
 * ILI 9488 REGS
*********************/

/* Level 1 Commands (from the display Datasheet) */
#define ILI9488_CMD_NOP                             0x00
#define ILI9488_CMD_SOFTWARE_RESET                  0x01
#define ILI9488_CMD_READ_DISP_ID                    0x04
#define ILI9488_CMD_READ_ERROR_DSI                  0x05
#define ILI9488_CMD_READ_DISP_STATUS                0x09
#define ILI9488_CMD_READ_DISP_POWER_MODE            0x0A
#define ILI9488_CMD_READ_DISP_MADCTRL               0x0B
#define ILI9488_CMD_READ_DISP_PIXEL_FORMAT          0x0C
#define ILI9488_CMD_READ_DISP_IMAGE_MODE            0x0D
#define ILI9488_CMD_READ_DISP_SIGNAL_MODE           0x0E
#define ILI9488_CMD_READ_DISP_SELF_DIAGNOSTIC       0x0F
#define ILI9488_CMD_ENTER_SLEEP_MODE                0x10
#define ILI9488_CMD_SLEEP_OUT                       0x11
#define ILI9488_CMD_PARTIAL_MODE_ON                 0x12
#define ILI9488_CMD_NORMAL_DISP_MODE_ON             0x13
#define ILI9488_CMD_DISP_INVERSION_OFF              0x20
#define ILI9488_CMD_DISP_INVERSION_ON               0x21
#define ILI9488_CMD_PIXEL_OFF                       0x22
#define ILI9488_CMD_PIXEL_ON                        0x23
#define ILI9488_CMD_DISPLAY_OFF                     0x28
#define ILI9488_CMD_DISPLAY_ON                      0x29
#define ILI9488_CMD_COLUMN_ADDRESS_SET              0x2A
#define ILI9488_CMD_PAGE_ADDRESS_SET                0x2B
#define ILI9488_CMD_MEMORY_WRITE                    0x2C
#define ILI9488_CMD_MEMORY_READ                     0x2E
#define ILI9488_CMD_PARTIAL_AREA                    0x30
#define ILI9488_CMD_VERT_SCROLL_DEFINITION          0x33
#define ILI9488_CMD_TEARING_EFFECT_LINE_OFF         0x34
#define ILI9488_CMD_TEARING_EFFECT_LINE_ON          0x35
#define ILI9488_CMD_MEMORY_ACCESS_CONTROL           0x36
#define ILI9488_CMD_VERT_SCROLL_START_ADDRESS       0x37
#define ILI9488_CMD_IDLE_MODE_OFF                   0x38
#define ILI9488_CMD_IDLE_MODE_ON                    0x39
#define ILI9488_CMD_COLMOD_PIXEL_FORMAT_SET         0x3A
#define ILI9488_CMD_WRITE_MEMORY_CONTINUE           0x3C
#define ILI9488_CMD_READ_MEMORY_CONTINUE            0x3E
#define ILI9488_CMD_SET_TEAR_SCANLINE               0x44
#define ILI9488_CMD_GET_SCANLINE                    0x45
#define ILI9488_CMD_WRITE_DISPLAY_BRIGHTNESS        0x51
#define ILI9488_CMD_READ_DISPLAY_BRIGHTNESS         0x52
#define ILI9488_CMD_WRITE_CTRL_DISPLAY              0x53
#define ILI9488_CMD_READ_CTRL_DISPLAY               0x54
#define ILI9488_CMD_WRITE_CONTENT_ADAPT_BRIGHTNESS  0x55
#define ILI9488_CMD_READ_CONTENT_ADAPT_BRIGHTNESS   0x56
#define ILI9488_CMD_WRITE_MIN_CAB_LEVEL             0x5E
#define ILI9488_CMD_READ_MIN_CAB_LEVEL              0x5F
#define ILI9488_CMD_READ_ABC_SELF_DIAG_RES          0x68
#define ILI9488_CMD_READ_ID1                        0xDA
#define ILI9488_CMD_READ_ID2                        0xDB
#define ILI9488_CMD_READ_ID3                        0xDC

/* Level 2 Commands (from the display Datasheet) */
#define ILI9488_CMD_INTERFACE_MODE_CONTROL          0xB0
#define ILI9488_CMD_FRAME_RATE_CONTROL_NORMAL       0xB1
#define ILI9488_CMD_FRAME_RATE_CONTROL_IDLE_8COLOR  0xB2
#define ILI9488_CMD_FRAME_RATE_CONTROL_PARTIAL      0xB3
[... 112 lines cut ...]

	/*Memory write*/
	ili9341_send_cmd(0x2C);

	ili9341_send_color((void*)mybuf, size * 3);

	 heap_caps_free(mybuf);
}

```

At disp_spi_init replace this:

 

```
  spi_bus_config_t buscfg={
            .miso_io_num=-1,
            .mosi_io_num=DISP_SPI_MOSI,
            .sclk_io_num=DISP_SPI_CLK,
            .quadwp_io_num=-1,
            .quadhd_io_num=-1,
            .max_transfer_sz = DISP_BUF_SIZE * 3,
    };
```

Mat-Alm, 2020-01-23:
[image]

C47D, 2020-01-24:
Hi Mat,

Great to see it working, I was working on the support on ILI9488 on [this](https://github.com/littlevgl/lv_port_esp32_ili9341/pull/44) pull request, i will check your changes and incorporate what i was missing.

Thanks for the help :)

C47D, 2020-01-25:
Updated the pull request with your code @Mat-Alm, i hope we're near to make it work!

kisvegabor, 2020-02-07:
@Mat-Alm could you test @C47D's version [here](https://github.com/littlevgl/lv_port_esp32_ili9341/pull/44)?

Mat-Alm, 2020-02-07:
> @Mat-Alm could you test @C47D's version [here](https://github.com/littlevgl/lv_port_esp32_ili9341/pull/44)?

How to download/clone the full project with this changes?

C47D, 2020-02-07:
@Mat-Alm , you need to clone my repo and track the `ili9488_support` branch, something like this:

```
$ git clone --recurse-submodules https://github.com/C47D/esp32_ili9341.git
$ cd esp32_ili9341
$ git checkout --track origin/ili9488_support
```

PR #44 (2020-02-10): [WORK IN PROGRESS] Add support for ILI9488 display controller
In order to support ILI9488 TFT display controller, fix #42
closed 2020-02-10 by kisvegabor