lvgl/lv_esp_idf #46: Choosing ESP32 DevKit v4 (38 GPIOS) as board doesn't allow the user to set custom pins for display control

C47D, 2020-02-09:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**
Choosing ESP32 DevKit v4 (38 GPIOS) as board doesn't allow the user to choose custom pins for the connection with the display.

**Code to reproduce the issue**
Choose ESP32 DevKit v4 (38 GPIOS) as board
[image]

**Expected Results**
Being able to choose custom pins for the connection between the microcontroller and the display.

**Actual Results**
The user can't set custom pins for the connection between the microcontroller and the display.

**ESP32 Chip version**
Any

**ESP-IDF version**
Any

**Development kit used**
ESP32 DevKit v4 (38 GPIOS)

**Development machine OS**
Any

**Compilation warnings/errors (if available)**

**If possible, copy the compilation log into a file and attach it here**

jeremyjh, 2020-02-09:
If you have that board nothing prevents you from simply using custom pin settings (although then you won't be using HSPI). Why would you want to pick pre-defined pins and then also pick the pins?

C47D, 2020-02-09:
Hi,

This issue was created to track this suggestion: https://github.com/littlevgl/lv_port_esp32_ili9341/pull/44#issuecomment-583782769

I guess that in case of wanting custom pins the user should choose the `None` option on the `Predefined board pinout`?

Maybe @Mat-Alm wants to choose different pins for the non SPI signals?

Mat-Alm, 2020-02-09:
> 
> 
> If you have that board nothing prevents you from simply using custom pin settings (although then you won't be using HSPI). Why would you want to pick pre-defined pins and then also pick the pins?

You are wrong!  Nothing prevent you to use any pins to "HSPI" or "VSPI". Yeah esp32 has many  bugs one of then is related with MISO line on HSPI but in this case the display does not use MISO :)

jeremyjh, 2020-02-09:
> In my case i have a debugger connected so I can't use the "standard" pinout.

In that case the correct choice in the menu is "None", because you do not want to use predefined pins.

C47D, 2020-02-10:
> > In my case i have a debugger connected so I can't use the "standard" pinout.
> 
> In that case the correct choice in the menu is "None", because you do not want to use predefined pins.

This is the solution

closed 2020-02-10 by C47D