lvgl/lv_esp_idf #47: [LV_CONF] Update LV_COLOR_DEPTH based on a menuconfig option

C47D, 2020-02-09:
It is now hardcoded, it should be a user configurable option when dealing with other driver configurations?

https://github.com/littlevgl/lv_port_esp32_ili9341/blob/edf5bf19083e911876318c0842c091a6dfdc6d24/components/lvgl/lv_conf.h#L28-L34

kisvegabor, 2020-02-10:
It can be updated like this:
```c
#ifndef LV_COLOR_DEPTH
#define LV_COLOR_DEPTH     16
#endif
```

C47D, 2020-02-10:
I think that what @Mat-Alm meant is being able to set the value to 1, 8, 16 or 32 using the menuconfig

coredump, 2020-02-26:
In a similar question:

Using menuconfig makes it almost never necessary to change stuff on the `lvgl.h` config, but still some stuff that is needed, like enabling themes and fonts. 

What would be an accepted way to change those settings without having to alter the lvgl or add all the possible options to menuconfig?

closed 2020-03-25 by stale[bot]