lvgl/lv_esp_idf #49: Rename of project (remove _ili9341)

seeul8er, 2020-02-13:
- Consider renaming the repository to just `lv_port_esp32` since ILI9488 display drivers are already supported and additional drivers are likely to be added in the future.

Keeping the current name might lead to the misconception that only the ili9341 display driver is supported.  

- Consider renaming the folder `/components/lvgl_ili9341` to something like `/components/lvgl_esp32_drivers`

Make it a general start-off project for all ESP32 + LittlevGL projects.

C47D, 2020-02-13:
I agree, inside `/components/lvgl_esp32_drivers` we could add a directory for the tft controllers and other for the touch controllers, something like so: `tft_controller` and `touch_controller`, do you agree @seeul8er?

kisvegabor, 2020-02-15:
I've just renamed to project.

I agree the with driver name changes. I usually use simply `tft` and `touch` but it doesn't matter too much. If you prefer `tft/touch_controller` it's also fine.

C47D, 2020-02-15:
Thanks for renaming the project, I will update the necessary files for the new directory structure.

closed 2020-02-17 by seeul8er