lvgl/lv_esp_idf #50: Init SPI bus in main and just the SPI device on TFT and touch controller

C47D, 2020-02-14:
Hi,

Right now we init the SPI bus and SPI devices on the TFT and touch init functions, i think is better to init the SPI bus on main so the user can add more devices, like sensors, etc, on main instead of modifying the TFT and touch controller functions to do so.

kisvegabor, 2020-02-15:
> I think is better to init the SPI bus on main so the user can add more devices, like sensors, etc, on main instead of modifying the TFT and touch controller functions to do so, maybe something like this:

I agree. It'd be really clearer this way.

C47D, 2020-02-15:
Great, I will do some tests to implement it. I think @jeremyjh have something similar in his repo, any suggestions?

C47D, 2020-02-29:
Closed with #55

closed 2020-02-29 by C47D