lvgl/lv_esp_idf #52: Add examples using ESP32 peripherals

C47D, 2020-02-20:
As #39 explains there are certain consideratio´n when using some ESP32 peripherals, how can we add examples using some peripherals? Maybe having the lvgl and other example projects and choosing one at configuration time?

Do you know if this is possible @jeremyjh?

jeremyjh, 2020-02-21:
There is more than one way to do it, but yeah one possibility would be to have a separate main for each example that are not added to the build files. Instead the project has a single main.c that includes the correct main file for the chose example based on those CONFIG defines, e.g. 

```c
#ifdef CONFIG_LVGL_EXAMPLE_WIFI 
#include "examples/wifi_main.c"
#endif
```

You don't want this configuration prompt in the library because the library doesn't use the examples, but you can have a Kconfig file in the main folder. I did that in the loboris driver fork for the settings that apply only to the example vs the library:
https://github.com/jeremyjh/ESP32_TFT_library/blob/master/main/Kconfig.projbuild

C47D, 2020-03-13:
go away

kisvegabor, 2020-03-13:
I've added the pinned label to not mark it as stale.