lvgl/lv_esp_idf #60: Add driver for monochrome display

C47D, 2020-02-29:
To use OLED or any other kind of monochrome display we need to edit the `lv_conf.h` file and set `LV_COLOR_DEPTH` to 1. Also edit the `disp_flush` function.

C47D, 2020-04-24:
There's a display driver for the SSD1306 chip.

X-Ryl669, 2020-04-29:
Any plan for SSD1327 ?

C47D, 2020-04-29:
@X-Ryl669 Not at the moment, I don't have any display with that driver, there's support for the SSD1306, if it's similar we can get SSD1327 working.

Do you have any display with that driver?

X-Ryl669, 2020-04-30:
Yes, it's very similar. It's 128x128, 4 bits per pixel (instead of just 128x64 and 1 bit/per pixel for SSD1306), but it's the same I2C address, almost the same command words. The initialization is different through:
Instead of:
```
static const PROGMEM uint8_t s_SSD1306_lcd128x32_initData[] =
{
    0xAE, 0x00,          // display off
    0xD5, 0x01, 0x80,    // Clock div
    0xA8, 0x01, 31,      // Set multiplex
    0xD3, 0x01, 0x00,    // --no offset
    0x40, 0x00,          // Set display offset
    0x8D, 0x01, 0x14,    // Set charge pump
    0xA0| 0x01, 0x00,    // Reverse mapping
    0xC8, 0x00,          // Decrement
    0xDA, 0x01, 0x02,    // Set com pins
    0x81, 0x01, 0x7F,    // contast value
    0xD9, 0x01, 0x22,    // 0x1F Precharge
    0xDB, 0x01, 0x40,    // Precharge
    0x20, 0x01, 0x00,    // Set horizontal addressing mode
    0xA4, 0x00,          // Display resume
    0xA6, 0x00,          // Normal display
    0xAF, 0x00,          // Display on
};
```
you should have:
```
static const PROGMEM uint8_t s_SSD1327_lcd128x128_initData[] =
{
    0xFD, 0x01, 0x12,  // Unlock OLED
    0xAE, 0x00,        // OFF                         /* display off */
    0xA8, 0x01, 0x7F,  // multiplex 128
    0xA1, 0x01, 0x00,  // Start line
    0xA2, 0x01, 0x00,  // Display offset
    0xA0, 0x01, 0x40 | 0x10 | 0x00 | (0x02 | 0x01),   // REMAP: horizontal increment mode
    0xAB, 0x01, 0x01,  // VDD internal
    0x81, 0x01, 0x70,  // CONTRAST
    0xB1, 0x01, 0x55,  // PHASE 0x51
    0xB3, 0x01, 0x01,  // CLK
    0xBC, 0x01, 0x08,  // PRECHARGE
    0xBE, 0x01, 0x07,  // VCOMH voltage
    0xB6, 0x01, 0x01,  // Second pre-charge
    0xA4, 0x00,        // NORMAL
    0x2E, 0x00,        // Deactivate scroll
    0xAF, 0x00,        // Display ON
};
```
Starting a block is done this way:
```
    this->send(0x15);
    this->send(x / 2);
    this->send( (rx < width) ? rx : (width - 1)) / 2 );
    this->send(0x75);
    this->send(y);
    this->send(height - 1);
    this->send(0x40);
```
instead of 
```
    this->send(0x21); // column addr
    this->send(x);
    this->send(w ? (x + w - 1) : (width - 1));
    this->send(0x22); // page addr
    this->send(y);
    this->send((height >> 3) - 1);
    this->send(0x40);
```
I'm using ESP32 here with I2C mode, with this [module](https://fr.aliexpress.com/item/4000261580944.html?spm=a2g0s.12269583.0.0.30621114aBJrMQ)

C47D, 2020-04-30:
I will add support for it but I would like you to run the tests, can you?.

maizezoidberg, 2021-07-02:
Hello guys! Have any news about SSD1327? I have display (128x128) with this controller and I can help with testing driver support

C47D, 2021-07-06:
Hi @maizezoidberg I have not, could you please open a new issue to track the progress on it?