lvgl/lv_esp_idf #62: config dosent working

app-z, 2020-03-02:
Something wrong with config

$ ls -la
-rw-rw-r--  1 appz appz 15843 Mar  2 22:57 sdkconfig
-rw-rw-r--  1 appz appz 27199 Mar  2 22:55 sdkconfig.old

$ make menuconfig

I cant see items menu connected with Sample Littlevgl. Therefore compilation is fail:

> In file included from /home/appz/esp/lv_port_esp32/components/lvgl/lvgl/lvgl.h:26:0,
>                  from /home/appz/esp/lv_port_esp32/components/lv_examples/lv_examples/lv_apps/benchmark/benchmark_bg.c:1:
> /home/appz/esp/lv_port_esp32/components/lvgl/lvgl/src/lv_core/lv_obj.h:39:2: error: #error "LittlevGL: LV_HOR_RES_MAX and LV_VER_RES_MAX must be greater than 0"
>  #error "LittlevGL: LV_HOR_RES_MAX and LV_VER_RES_MAX must be greater than 0"
>   ^
> CC build/libsodium/libsodium/src/libsodium/crypto_auth/hmacsha256/auth_hmacsha256.o
> In file included from /home/appz/esp/lv_port_esp32/components/lvgl/lvgl/src/lv_core/lv_disp.h:17:0,
>                  from /home/appz/esp/lv_port_esp32/components/lvgl/lvgl/src/lv_core/lv_disp.c:9:
> /home/appz/esp/lv_port_esp32/components/lvgl/lvgl/src/lv_core/lv_obj.h:39:2: error: #error "LittlevGL: LV_HOR_RES_MAX and LV_VER_RES_MAX must be greater than 0"
>  #error "LittlevGL: LV_HOR_RES_MAX and LV_VER_RES_MAX must be greater than 0"
>

C47D, 2020-03-02:
Hi,

How are you cloning the repository? Just tried myself and had no issues, here's what i did:
```
$ git clone --recurse-submodules https://github.com/littlevgl/lv_port_esp32.git
...
$ cd lv_port_esp32
$ idf.py menuconfig
$ idf.py build
```

What values are you configuring `TFT display width in pixels` and `TFT display heigth in pixels`?

C47D, 2020-03-03:
Fixed with #64 

[image]

closed 2020-03-03 by C47D