lvgl/lv_esp_idf #66: Driver ILI9488 renders only a portion of the screen

Djaison, 2020-03-03:
**Describe the issue**
Hey. I place my question here, because I think that the error is due to driver ILI9488.
I ported the project to ESP32 according to the readme. I use Waveshare_ILI9486 display. After compiling and downloading, I see on the screen an incomplete picture, besides an inverted 90 degrees. Screen resolution 320x480, this parameters I save to sdkconfig, all other parameters I use the default. SPI frequency = 10 MHz, I use HSPI, but at higher frequencies there is no picture at all, perhaps because of the wired connection. With what may be due to this problem?

**ESP-IDF version**
last master

**Development kit used**
ESP-WROOM32

**Development machine OS**
Windows

**If possible, copy the compilation log into a file and attach it here**
Info: lv_init ready     (../components/lvgl/lvgl/src/lv_core/lv_obj.c #117)
ILI9488 initialization.
Enabling backlight.
Info: Screen create ready       (../components/lvgl/lvgl/src/lv_core/lv_obj.c #211)
Info: Screen create ready       (../components/lvgl/lvgl/src/lv_core/lv_obj.c #211)
Info: Screen create ready       (../components/lvgl/lvgl/src/lv_core/lv_obj.c #211)
Info: image created     (../components/lvgl/lvgl/src/lv_objx/lv_img.c #108)
Info: lv_img_set_src:  `LV_IMG_SRC_VARIABLE` type found         (../components/lvgl/lvgl/src/lv_objx/lv_img.c #152)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: button matrix created     (../components/lvgl/lvgl/src/lv_objx/lv_btnm.c #127)
Info: tab view created  (../components/lvgl/lvgl/src/lv_objx/lv_tabview.c #184)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: page created      (../components/lvgl/lvgl/src/lv_objx/lv_page.c #167)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: page created      (../components/lvgl/lvgl/src/lv_objx/lv_page.c #167)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: page created      (../components/lvgl/lvgl/src/lv_objx/lv_page.c #167)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: page created      (../components/lvgl/lvgl/src/lv_objx/lv_page.c #167)
Info: label created     (../components/lvgl/lvgl/src/lv_objx/lv_label.c #154)
Info: text area created         (../components/lvgl/lvgl/src/lv_objx/lv_ta.c #193)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: page created      (../components/lvgl/lvgl/src/lv_objx/lv_page.c #167)
Info: list created      (../components/lvgl/lvgl/src/lv_objx/lv_list.c #148)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: button created    (../components/lvgl/lvgl/src/lv_objx/lv_btn.c #145)
Info: image created     (../components/lvgl/lvgl/src/lv_objx/lv_img.c #108)
Info: label created     (../components/lvgl/lvgl/src/lv_objx/lv_label.c #154)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: button created    (../components/lvgl/lvgl/src/lv_objx/lv_btn.c #145)
Info: image created     (../components/lvgl/lvgl/src/lv_objx/lv_img.c #108)
Info: label created     (../components/lvgl/lvgl/src/lv_objx/lv_label.c #154)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: button created    (../components/lvgl/lvgl/src/lv_objx/lv_btn.c #145)
Info: image created     (../components/lvgl/lvgl/src/lv_objx/lv_img.c #108)
Info: label created     (../components/lvgl/lvgl/src/lv_objx/lv_label.c #154)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: button created    (../components/lvgl/lvgl/src/lv_objx/lv_btn.c #145)
Info: image created     (../components/lvgl/lvgl/src/lv_objx/lv_img.c #108)
Info: label created     (../components/lvgl/lvgl/src/lv_objx/lv_label.c #154)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: button created    (../components/lvgl/lvgl/src/lv_objx/lv_btn.c #145)
Info: image created     (../components/lvgl/lvgl/src/lv_objx/lv_img.c #108)
Info: label created     (../components/lvgl/lvgl/src/lv_objx/lv_label.c #154)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: button created    (../components/lvgl/lvgl/src/lv_objx/lv_btn.c #145)
Info: image created     (../components/lvgl/lvgl/src/lv_objx/lv_img.c #108)
Info: label created     (../components/lvgl/lvgl/src/lv_objx/lv_label.c #154)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: button created    (../components/lvgl/lvgl/src/lv_objx/lv_btn.c #145)
Info: image created     (../components/lvgl/lvgl/src/lv_objx/lv_img.c #108)
Info: label created     (../components/lvgl/lvgl/src/lv_objx/lv_label.c #154)
Info: container created         (../components/lvgl/lvgl/src/lv_objx/lv_cont.c #113)
Info: label created     (../components/lvgl/lvgl/src/lv_objx/lv_label.c #154)
Info: mesasge box created       (../components/lvgl/lvgl/src/lv_objx/lv_mbox.c #125)
Info: button matrix created     (../components/lvgl/lvgl/src/lv_objx/lv_btnm.c #127)
Info: chart created     (../components/lvgl/lvgl/src/lv_objx/lv_chart.c #140)
Info: bar created       (../components/lvgl/lvgl/src/lv_objx/lv_bar.c #116)
Info: slider created    (../components/lvgl/lvgl/src/lv_objx/lv_slider.c #104)
Info: image draw: cache miss, cached to an empty entry  (../components/lvgl/lvgl/src/lv_draw/lv_img_cache.c #109)

[image]

C47D, 2020-03-03:
Hi,

> Hey. I place my question here, because I think that the error is due to driver ILI9488.
> I ported the project to ESP32 according to the readme. I use Waveshare_ILI9486 display. 

Is your display using the ILI9488 or ILI9486 controller? Right now the repo only supports ILI9341, ILI9488, ST7789 and soon HX8357D. The configuration might be different from ILI9486 to ILI9488.

Have you tested your display with other libraries? Maybe using the ESP32 as Arduino, this way we can be sure the display is working properly.

Regards

Djaison, 2020-03-03:
Yes, my display still has ILI9486 controller, but I decided that it should work with the driver ILI9488, since some resources on the Internet, he is referred to as ILI9486 / ILI9488. Probably knows the exact answer ILI9488 driver developer. This display exactly working, I used it with a library from Arduino Waveshare on ArduinoMega. Thanks for the quick response!

Regards

C47D, 2020-03-03:
Ok, I will take a look at the code, last weekend i was using the ILI9488 and had no issues.

Djaison, 2020-03-04:
Pins are configured as follow
[image]

Djaison, 2020-03-04:
Could you provide your project sdkconfig on ili9488? I use the demo example, maybe the problem is in the wrong dma configuration?

C47D, 2020-03-04:
Maybe your repo clone is somewhat old, the repo has been renamed to lv_port_esp32.

You can get the latest repo with:
```
$ git clone --recurse-submodules https://github.com/littlevgl/lv_port_esp32.git
$ cd lv_port_esp32
$ idf.py menuconfig
/* Go to Components->Littlevgl (LVGL) TFT display controller->Select a display controller model and choose ILI9488 */
/* Then set your pin assignments */
```

Djaison, 2020-03-05:
The problem is similar to the display. I use this. 
[image]
Most likely the problem is spi converter, which is located on the reverse side, maybe he realizes not all the functionality. Successfully launched with other display who has a ST7789 driver. Sorry to bother you, I think the question can be closed, thanks again for your help!

closed 2020-03-05 by Djaison