lvgl/lv_esp_idf #70: ILI9341 and XPT2046 on same SPI

ndunello, 2020-03-06:
My HW connection is as follows and use use idf 4.0:
**Diplay**:

- MISO 25
- MOSI 23
- CLK 19
- CS 22
- DC 21
- RESET 18
- BACKLIGHT 5

**Touch**

- CS 32
- IRQ 33

I found a problem in xpt2046_init function in xpt2046.c file.
I modified
`.pin_bit_mask = 1UL << XPT2046_IRQ,`
to
`.pin_bit_mask = 1ULL << XPT2046_IRQ,`
however the touch does not work.
Note that I have already tested it with the esp-iot-solution (IDF 3.x) lvgl_thermostat project.

C47D, 2020-03-06:
Hi @ndunello,

Thanks for the report, I will try to replicate it later today, were you using 30cc022?. What SPI were you using?

Regards

ndunello, 2020-03-06:
Hi @C47D,
I use HSPI_HOST.
About "30cc022", forgive me, but I can't answer. how do i get it?
Yesterday I downloaded master "zip" file from the repository.

Regards

C47D, 2020-03-06:
> About "30cc022", forgive me, but I can't answer. how do i get it?
> Yesterday I downloaded master "zip" file from the repository.

30cc022 is the commit you're working on, if you downloaded master zip file then yes, you're using 30cc022 :), please let me test it later today and see why the touch controller isn't working.

C47D, 2020-03-09:
> I found a problem in xpt2046_init function in xpt2046.c file.
> I modified
> .pin_bit_mask = 1UL << XPT2046_IRQ,
> to
> .pin_bit_mask = 1ULL << XPT2046_IRQ,

This issue is solved with 5dfc897.

I wired up the ILI9341 based display I have at hand and couldn´t make the IRQ pin signal of the touch sensor go low when I touch the screen, it only worked once and then I couldn't get it working again, do you also have this issue? Is it a known problem?

bzgec, 2020-03-09:
> I wired up the ILI9341 based display I have at hand and couldn´t make the IRQ pin signal of the touch sensor go low when I touch the screen, it only worked once and then I couldn't get it working again, do you also have this issue? Is it a known problem?

Same here. 
In my case it only works in this sequence:
- board must be disconnected for few seconds (around 30s)
- connect the board to computer (do not touch display)
- flash new program to the board (even if there are no changes)
- everything works well  

But if you reset the board or disconnect the power and than reconnect it again it works only for one click.

C47D, 2020-03-09:
Hi,

I will try your sequence, yesterday I ended up searching the problem on arduino forums and found nothing related, do you think this is an issue with the touch controller or maybe we're configuring wrong the IRQ pin on the esp32 side?

bzgec, 2020-03-09:
Commit `0de169d` works fine for me.
I am still trying to figure out what could be the cause of the problem.

C47D, 2020-03-09:
I can try bisecting until find the faulty commit.

bzgec, 2020-03-09:
I think I found the solution with the help of logic analyzer. The problem was in `tp_spi.c` file. SPI configuration was set to full duplex mode and the attempt to set it to `SPI_DEVICE_HALFDUPLEX` in `spi_transaction_t` instead in `spi_device_interface_config_t`.
The two functions i changed.
```
void tp_spi_add_device(spi_host_device_t host)
{
	spi_device_interface_config_t devcfg={
#if CONFIG_LVGL_TOUCH_CONTROLLER == TOUCH_CONTROLLER_STMPE610
		.clock_speed_hz=1*1000*1000,           //Clock out at 1 MHz
		.mode=1,                               //SPI mode 1
#else
		.clock_speed_hz=2*1000*1000,           //Clock out at 2 MHz
		.mode=0,                               //SPI mode 0
#endif
		.spics_io_num=TP_SPI_CS,               //CS pin
		.queue_size=1,
		.pre_cb=NULL,
		.post_cb=NULL,
        .command_bits = 8,
        .address_bits = 0,
        .dummy_bits = 0,
        .flags = SPI_DEVICE_HALFDUPLEX | SPI_DEVICE_NO_DUMMY,
	};
	
	//Attach the Touch controller to the SPI bus
	tp_spi_add_device_config(host, &devcfg);
}
```
and
```
void tp_spi_read_reg(uint8_t reg, uint8_t* data, uint8_t byte_count)
{
	spi_transaction_t t;
	// spi_transaction_ext_t et;
	
	memset(&t, 0, sizeof(t));
	    
	// Read - send first byte as command
	t.length = byte_count * 8 + sizeof(reg);
	t.rxlength = byte_count * 8;
	t.cmd = reg;
	t.rx_buffer = data;
	t.flags = 0;
	esp_err_t ret = spi_device_transmit(spi, &t);
	assert(ret == ESP_OK);
}
```

I noticed that `tp_spi_write_reg()` function also has this error.

C47D, 2020-03-10:
@bzgec I've added your modifications to this branch https://github.com/littlevgl/lv_port_esp32/tree/fix_70 could you test it? I didn't had time to solve the IRQ pin working only once.

bzgec, 2020-03-10:
I was too fast and made a mistake.

`t.length = byte_count * 8 + sizeof(reg);` 
should be changed to
`t.length = (byte_count + sizeof(reg)) * 8;`

ndunello, 2020-03-10:
I just tried https://github.com/littlevgl/lv_port_esp32/tree/fix_70 and I can see the print of the coordinates on the console.
Now, if i touch on "Write", "List" and "Chart" tab nothing is happening. Its coordinates are (approximately in the center):
"Write":     x = 104, y = 191
"List":    x = 73, y = 188
"Chart":    x = 45, y = 190
did I make any mistakes in the configuration?
The touch display I use is [this](http://www.lcdwiki.com/2.8inch_SPI_Module_ILI9341_SKU:MSP2807)

bzgec, 2020-03-10:
I created the pull request to fix this:
> I was too fast and made a mistake.
`t.length = byte_count * 8 + sizeof(reg);`
should be changed to
`t.length = (byte_count + sizeof(reg)) * 8;`

On branch `fix_70`, it is working ok for me, but i did not test for shared SPI.

bzgec, 2020-03-10:
> "Write": x = 104, y = 191
"List": x = 73, y = 188
"Chart": x = 45, y = 190
did I make any mistakes in the configuration?

I guess that you need to change `Touchpanel Configuration`.
Here is my configuration (note that i have 3.2 inch display):
[image]

ndunello, 2020-03-10:
OK, works "almost" well.
the last problem is in partial display, as in the figure.
[image]
Display Configuration is
[image]
Touch Configurazion is
[image]
@bzgec what resolution does your 3.2 inch display have?

pratikmokashi, 2020-03-10:
@ndunello You have wrong configuration in the display
correct setting :- 
(320) TFT display width in pixels.
(240) TFT display height in pixels.

ndunello, 2020-03-10:
thank @pratikmokashi, you are perfectly right!
Now everything works fine
Thank you all

C47D, 2020-03-10:
@bzgec can you test it with shared SPI buses?

ndunello, 2020-03-10:
@C47D:
 - yes, touch working properly!
 - yes, I use same pinout as my first post (SHARED_SPI_BUS)

pratikmokashi, 2020-03-10:
@ndunello can you help me how to use the touch with this display?
And all the setting you used for it work

C47D, 2020-03-10:
@bzgec @pratikmokashi @ndunello thanks everyone for the help to close this issue :)

PR #74 (2020-03-10): Fix SPI transfer mode when reading and writing to touch controller 
Closes #70
closed 2020-03-10 by C47D