lvgl/lv_esp_idf #77: "Install this project as a library submodule in your own project" not working

eddeyonline, 2020-03-13:
When following instructions to add lv_port_esp32 with CMake 4.0 repeated CMake error:

-- Building ESP-IDF components for target esp32
CMake Error at C:/ESP/esp-idf/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'lvgl_tft'.
Call Stack (most recent call first):
  C:/ESP/esp-idf/tools/cmake/build.cmake:211 (__build_resolve_and_add_req)
  C:/ESP/esp-idf/tools/cmake/build.cmake:425 (__build_expand_requirements)
  C:/ESP/esp-idf/tools/cmake/project.cmake:337 (idf_build_process)
  CMakeLists.txt:7 (project)

Fixed by adding additional EXTRA_COMPONENT_DIRS arguments as follows:

set(EXTRA_COMPONENT_DIRS externals/lv_port_esp32/components externals/lv_port_esp32/components/lvgl_esp32_drivers externals/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_touch externals/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft)

Could this be because lvgl_touch and lvgl_tft are within a subdirectory of components rather than within components directory?

Hope this is helpful!

C47D, 2020-03-14:
Hi,

Thanks for reporting the issue, they're already added into the CMakeLists.txt in the root of the repo
```
set(EXTRA_COMPONENT_DIRS components/lvgl_esp32_drivers components/lvgl_esp32_drivers/lvgl_touch components/lvgl_esp32_drivers/lvgl_tft)
```

Where did you added it?

eddeyonline, 2020-03-14:
It needs to be modified in the instructions in readme.md under the subheading:
"Install this project as a library submodule in your own project"

There it states:
The project root CMakeLists.txt file needs one line added, just before the project to add the extra components directory to the path like this:

#CMakeLists.txt
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

set(EXTRA_COMPONENT_DIRS externals/lv_port_esp32/components)

project(blink)

But I found I needed to add the following:

set(EXTRA_COMPONENT_DIRS externals/lv_port_esp32/components externals/lv_port_esp32/components/lvgl_esp32_drivers externals/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_touch externals/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft)

I'm still having other issues trying to work out how to incorporate littlegvl into my own projects...

C47D, 2020-03-14:
Thanks for the clarification, I will update it asap.

Maybe you can try use this as your project template, just configuring it with menuconfig and commenting out the `demo_create` line on the main.c file.

All the feedback is welcome, we been adding some TFT drivers and touch controllers but I don't have them all to test. What driver are you planning to use?

eddeyonline, 2020-03-14:
Thanks, I'll try commenting out the demo_create line and adding in my own function there. 

I'm using a 2.4" 240x320 TFT with an ILI9341 display driver and XPT2046 touch driver, so pretty standard. Using the demo file it worked great. I just needed to make sure I hooked it up as per the readme.md file and deselect "invert Y axis" in menuconfig to get the demo to work perfectly. Thanks for your great work with the library! I'll let you know how I go getting my own code to run.

eddeyonline, 2020-03-14:
I've been trying out a few of the apps and tutorials and have had some success with some and run into issues with others. For all of them I've used the demo template and have changed the include to the appropriate *.h file, changed the name of the function called in app_main() and adjusted the variables in lv_ex_conf as appropriate. I found the following worked fine:
- demo, with and without slideshow
- benchmark
- lv_test_group
- lv_test_obj

Whichever tutorial I tried to build it failed with the following error:

./components/lv_examples/lv_examples/lv_tutorial/8_animations/lv_tutorial_animations.c:87:6: error: 'lv_anim_t' {aka 'struct _lv_anim_t'} has no member named 'user_data'
     a.user_data = NULL;

Commenting out this line resulted in all the tutuorials I tried working fine, though the images one seemed to be missing some source files.

sysmon failed with the following error:
../components/lv_examples/lv_examples/lv_apps/sysmon/sysmon.c: In function 'sysmon_task':
../components/lv_examples/lv_examples/lv_apps/sysmon/sysmon.c:135:13: error: passing argument 1 to restrict-qualified parameter aliases with argument 3 [-Werror=restrict]
     sprintf(buf_long, "%s"LV_TXT_COLOR_CMD"%s MEMORY: %d %%"LV_TXT_COLOR_CMD"\n"
             ^~~~~~~~
../components/lv_examples/lv_examples/lv_apps/sysmon/sysmon.c:140:13:
             buf_long,
             ~~~~~~~~

tpcal failed with following error:
../main/main.c: In function 'app_main':
../main/main.c:115:2: error: implicit declaration of function 'tpcal_create'; did you mean 'lv_arc_create'? [-Werror=implicit-function-declaration]
  tpcal_create();
  ^~~~~~~~~~~~

I double checked ex_conf and the include, so I'm not sure what the issue was here. I got a similar issue with terminal, error message below:

When attempting to use terminal:
../main/main.c: In function 'app_main':
../main/main.c:115:2: error: implicit declaration of function 'terminal_create'; did you mean 'lv_win_create'? [-Werror=implicit-function-declaration]
  terminal_create();
  ^~~~~~~~~~~~~~~

For anyone using the same touchscreen as me, the variables that resulted in the good calibration are as follows:
CONFIG_LVGL_TOUCH_X_MIN=200
CONFIG_LVGL_TOUCH_Y_MIN=120
CONFIG_LVGL_TOUCH_X_MAX=1900
CONFIG_LVGL_TOUCH_Y_MAX=1900
CONFIG_LVGL_TOUCH_INVERT_X=y
# CONFIG_LVGL_TOUCH_INVERT_Y is not set

C47D, 2020-03-14:
Thanks for the update, does the tutorial you're following need or is based on a specific version of lvgl? As far as I know the version used in this project is somewhat old. Maybe that's why some lvgl types aren't available.

C47D, 2020-03-16:
With #80 merge you should be able to compile and run the sysmon application, I'm still trying to figure out how to run terminal and tpcal apps.

eddeyonline, 2020-03-17:
Thanks Carlos!

I'll give it a go :)

C47D, 2020-03-17:
Sure np problem, I was testing the applications with a ST7789 based display with no touch controller so I didn't tried to run the tpcal application, but I will try with a similar display to yours next weekend (or before if my day job allows it).

I wasn't able to fully understand the terminal application, let me know if I'm wrong but I understand it as it display the data received by the microcontroller via UART?

C47D, 2020-03-19:
@eddeyonline just updated the instructions on the README about using this repo on user projects, let me know if it works for you.

closed 2020-04-17 by stale[bot]