lvgl/lv_esp_idf #79: Readme.md should be enhanced regarding adjustments in Makefile

an-erd, 2020-03-16:
The Readme says in section [Readme.md#makefile](https://github.com/littlevgl/lv_port_esp32#makefile) to add the following line(s):

> EXTRA_COMPONENT_DIRS := externals/lv_port_esp32/components

If only this directory is added, the Kconfig files in the following files are not reflected in `menuconfig`:
> ./externals/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft/Kconfig
> ./externals/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_touch/Kconfig

and thus compilation does not work.

Adding the following line works for me:

> EXTRA_COMPONENT_DIRS := externals/lv_port_esp32/components externals/lv_port_esp32/components/lvgl_esp32_drivers externals/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_touch externals/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_tft

C47D, 2020-03-16:
Hi @an-erd,

You're right, the README haven't been totally updated as should be. As user what do you find the most easy way to use this project as a template for your personal project:
- By doing what you did based on the instructions of https://github.com/littlevgl/lv_port_esp32#install-this-project-as-a-library-submodule-in-your-own-project
- Deleting the demo_create function call and add your application like any other esp32 idf based project?

I ask because #77 have the same problem

an-erd, 2020-03-17:
Hi, sorry I missed the point in #77. Please see my comment in #81,

an-erd, 2020-03-17:
I'm using ESP-IDF with LittlevGL as a submodule, so your first option. I was asking why you put this in externals and not components? In components I've all other submodules, so if there's not a reason for that I'd propose to move to submodules.

So far with the above adjustment in the  makefile it's working pretty good, integration was very easy and straightforward. I currently only have a M5Stack with an ILI9341 without a touchpad. I'm using this code called from main.c I took from the examples, and doing the work in  `lv_testdisplay_create();` :

>void initialize_lv()
>{
>    lv_init();
>    lvgl_driver_init();
>    static lv_color_t buf1[DISP_BUF_SIZE];
>    static lv_color_t buf2[DISP_BUF_SIZE];
>    static lv_disp_buf_t disp_buf;
>    lv_disp_buf_init(&disp_buf, buf1, buf2, DISP_BUF_SIZE);
>    lv_disp_drv_t disp_drv;
>    lv_disp_drv_init(&disp_drv);
>    disp_drv.flush_cb = disp_driver_flush;
>    disp_drv.buffer = &disp_buf;
>    lv_disp_drv_register(&disp_drv);
>    esp_register_freertos_tick_hook(lv_tick_task);
>    lv_testdisplay_create();
>    while (1) {
>        vTaskDelay(1);
>        lv_task_handler();
>    }
>}
>

C47D, 2020-03-17:
> I'm using ESP-IDF with LittlevGL as a submodule, so your first option. I was asking why you put this in externals and not components? In components I've all other submodules, so if there's not a reason for that I'd propose to move to submodules.

I haven't updated the README as I should, should it be?
> EXTRA_COMPONENT_DIRS := components/lvgl_esp32_drivers\ components/lvgl_esp32_drivers/lvgl_touch\
components/lvgl_esp32_drivers/lvgl_tft\

If you see any other misleading thing on the README please comment it on the README, let get it up to date :)

C47D, 2020-03-18:
@an-erd Fixed in my side, let me know if it works for you.

closed 2020-04-13 by C47D