lvgl/lv_esp_idf #82: How make rotation screen (Set portrait orientation) Driver ILI9488

Djaison, 2020-03-17:
**ESP-IDF version**
last master

**Development kit used**
ESP-WROOM32

**Development machine OS**
Windows 10

Hello! How can I rotate the screen? I need to set portrait orientation.
I use display with driver ILI9488.

[image]

Djaison, 2020-03-17:
I made corrections to the void ili9488_init (void) function in the ili9488.с file, {ILI9488_CMD_MEMORY_ACCESS_CONTROL, {(0x20 | 0x08)}, 1} I changed it to {ILI9488_CMD_MEMORY_ACCESS_CONxRO} I can add 1 option, {Think} 1 menu} I’ll choose 1 menu}

C47D, 2020-03-17:
Hi @Djaison,

Can you try swapping the values on this config?
[image]

If that doesn't work i can add an option to the menuconfig as you've suggested.

Offtopic: Your interface looks really cool :)

Djaison, 2020-03-17:
> If that doesn't work i can add an option to the menuconfig as you've suggested.

it does not work :)

now it looks like this

[image]

C47D, 2020-03-17:
Ok, I will try to add an option to set the display on portrait mode from the menuconfig, but it will be later today.

an-erd, 2020-03-17:
It would be good to add this option for the ILI9341, too. I just worked on this today and found that for the M5Stack Fire in the file `externals\lv_port_esp32\components\lvgl_esp32_drivers\lvgl_tft\ili9341.c` writing 0x08 (instead of 0x68) to 0x36 worked for . With this the display shows in landscape, with the buttons on the bottom. I used this code:

> #if ILI9341_INVERT_DISPLAY
> 	uint8_t data[] = {0x08};
> 	// this same command also sets rotation (portrait/landscape) and inverts colors.
> 	// https://gist.github.com/motters/38a26a66020f674b6389063932048e4c#file-ili9844_defines-h-L24
> 	ili9341_send_cmd(0x36);
> 	ili9341_send_data(&data, 1);
> #endif

Value taken from ILI9341 datasheet, section 8.2.29. Memory Access Control (36h). Also height and width must be swapped, too. I did it like this in file `externals\lv_port_esp32\components\lvgl\lv_conf.h`
> #define LV_HOR_RES_MAX          (CONFIG_LVGL_DISPLAY_HEIGHT)
> #define LV_VER_RES_MAX          (CONFIG_LVGL_DISPLAY_WIDTH)

C47D, 2020-03-17:
> It would be good to add this option for the ILI9341, too.

Sure, I will add it later today, I'm at work right now.

Djaison, 2020-03-19:
I was able to configure the touch to the portrait orientation should also change the default settings in menuconfig, the operating settings such:
[image]

P.S. Touch driver is a FT6236

C47D, 2020-03-19:
It looks really nice :)

> I was able to configure the touch to the portrait orientation should also change the default settings in menuconfig, the operating settings such

We can add a note for that on the README, I don't know if it can be done automagically based on the display orientation.

Djaison, 2020-03-19:
it can be added to menuсonfig as a function of the orientation of the display, and later I will be able to provide it after checking

closed 2020-04-17 by stale[bot]