lvgl/lv_esp_idf #85: Install this project as a library submodule breaks build

Mair, 2020-03-24:
**Describe the issue**
[this commit](https://github.com/littlevgl/lv_port_esp32/commit/daa381851844123db770bc4402ab9b2058aff389#diff-04c6e90faac2675aa89e2176d2eec7d8) changed instructions on adding the library into the components folder as aposed to an external directory. Following these instructions breaks the build

## to reproduce...
1. create a new esp-idf project 
```
npx create-esp32-app
```
2. try to build
```
idf.py build
```
3. follow steps to add sub modules
```
mkdir -p components
git submodule add https://github.com/littlevgl/lv_port_esp32.git components/lv_port_esp32
git submodule update --init --recursive
```
4. clean the build
``
idf.py fullclean
``
5. build the project
```
idf.py build
```
build is now broken adding references in the CMakeLists.txt files doesn't make any difference 

**Expected Results**
`idf.py build` builds the project with no errors
**Actual Results**
cmake errors

**ESP-IDF version**
4.0

**Development machine OS**
windows
**Compilation warnings/errors (if available)**

**If possible, copy the compilation log into a file and attach it here**
C:\projects\spike\testv1>idf.py build
Checking Python dependencies...
Python requirements from C:\esp\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory c:\projects\spike\testv1\build
Executing "ninja all"...
[0/1] Re-running CMake...
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- Project version: 032159a-dirty
-- Building ESP-IDF components for target esp32
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.25.1.windows.1") 
CMake Error at C:/esp/esp-idf/tools/cmake/component.cmake:222 (message):
  CMake Warning (dev) at build_properties.temp.cmake:2 (set):

    Syntax error in cmake code at

      C:/projects/spike/testv1/build/build_properties.temp.cmake:2

    when parsing string

      C:\esp\tools\.espressif\python_env\idf4.0_py3.7_env\Scripts\python.exe

    Invalid escape sequence \e

    Policy CMP0010 is not set: Bad variable reference syntax is an error.  Run
    "cmake --help-policy CMP0010" for policy details.  Use the cmake_policy
    command to set the policy and suppress this warning.

  Call Stack (most recent call first):

    C:/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:1 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  CMake Error at
  C:/esp/tools/.espressif/tools/cmake/3.13.4/share/cmake-3.13/Modules/ExternalProject.cmake:980
  (define_property):

    define_property command is not scriptable

  Call Stack (most recent call first):

    C:/esp/esp-idf/tools/cmake/kconfig.cmake:1 (include)
    C:/esp/esp-idf/tools/cmake/idf.cmake:36 (include)
    C:/esp/esp-idf/tools/cmake/project.cmake:7 (include)
    C:/projects/spike/testv1/components/lv_port_esp32/CMakeLists.txt:3 (include)
    C:/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:60 (include)
    C:/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:76 (__component_get_requirements)

Call Stack (most recent call first):
  C:/esp/esp-idf/tools/cmake/build.cmake:404 (__component_get_requirements)
  C:/esp/esp-idf/tools/cmake/project.cmake:337 (idf_build_process)
  CMakeLists.txt:12 (project)

C47D, 2020-03-24:
Hi @Mair,

Thanks for the detailed report, I was able to reproduce the issue and did a quick fix. The source of the issue was that I forgot to add a note about the CMakeLists.txt on the lv_port_esp32 repo and also a typo in the main/CMakeLists.txt on your repo.

Here are the fixes, please let me know if it works for you so I can send a pull request to fix it.

Your application main/CMakeList.txt should look like this (the typo was **lvgl_examples**, it must be **lv_examples**):
```c
set (SOURCES main.c)

idf_component_register(SRCS ${SOURCES}
    INCLUDE_DIRS .
    REQUIRES lvgl_esp32_drivers lvgl lv_examples lvgl_tft lvgl_touch)

target_compile_definitions(${COMPONENT_LIB PRIVATE LV_CONF_INCLUDE_SIMPLE=1)
```

The file on components/lv_port_esp32/CMakeLists.txt should look like this:
```c
cmake_minimum_required(VERSION 3.5)

# include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(EXTRA_COMPONENT_DIRS components/lvgl_esp32_drivers components/lvgl_esp32_drivers/lvgl_touch components/lvgl_esp32_drivers/lvgl_tft)
# project(lvgl-demo)
```

With the include and project lines commented out.

Let me know if you have any issues after that changes, I'm able to build the demo application after it.

Regards
Carlos

Mair, 2020-03-24:
@C47D Legend!

> The file on components/lv_port_esp32/CMakeLists.txt should look like this:
```c++
cmake_minimum_required(VERSION 3.5)

# include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(EXTRA_COMPONENT_DIRS components/lvgl_esp32_drivers components/lvgl_esp32_drivers/lvgl_touch components/lvgl_esp32_drivers/lvgl_tft)
# project(lvgl-demo)
```
> With the include and project lines commented out.

Fixed this for me. Thanks!

C47D, 2020-03-24:
Can you test this?

```c
cmake_minimum_required(VERSION 3.5)

# include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(EXTRA_COMPONENT_DIRS components/lvgl_esp32_drivers components/lvgl_esp32_drivers/lvgl_touch components/lvgl_esp32_drivers/lvgl_tft)

if (NOT DEFINED PROJECT_NAME)
    project(lvgl-demo)
endif (NOT DEFINED PROJECT_NAME)
```

Mair, 2020-03-24:
In my case the above  compiles

C47D, 2020-03-24:
Ok, I will update the README with the following info
```c
cmake_minimum_required(VERSION 3.5)

# include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(EXTRA_COMPONENT_DIRS components/lvgl_esp32_drivers components/lvgl_esp32_drivers/lvgl_touch components/lvgl_esp32_drivers/lvgl_tft)

if (NOT DEFINED PROJECT_NAME)
    project(lvgl-demo)
endif (NOT DEFINED PROJECT_NAME)
```

Thanks for testing it tho, I will close the issue.

Regards

closed 2020-03-24 by C47D

Mair, 2020-03-24:
@C47D Just something to note...
ESP-IDF *should* automatically include all assets in the components folder which will make the cmakelits.txt file simpler. I'll test it out and let you know

Mair, 2020-03-24:
I was kind of hoping to get away with something like 
```c++
cmake_minimum_required(VERSION 3.5)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(menu_v2_poc)
```
and for main
```c++
set(COMPONENT_SRCS "main.c")
set(COMPONENT_ADD_INCLUDEDIRS "." )
register_component()
```
But due to the nested components folder the includes don't get picked up :(.  In any case happy to go with the above solution,
Thanks again

C47D, 2020-03-24:
Thanks for trying, I've just updated the README with the workaround.

maxd, 2020-10-01:
@C47D I want to clarify the following thing: you are using `if (NOT DEFINED PROJECT_NAME)` to disable call of `project(lvgl-demo)`. How about to use the same `if` to disable `include` directive with no need for make any manual changes (comment this line manually in sub-module repository)?

I have made this:

```
if (NOT DEFINED PROJECT_NAME)
    include($ENV{IDF_PATH}/tools/cmake/project.cmake)
endif (NOT DEFINED PROJECT_NAME)
```

and it solved the error:

```
...
CMake Error at /Users/mdobryakov/.esp-idf/tools/cmake/component.cmake:221 (message):
  CMake Error at
  /Applications/CLion.app/Contents/bin/cmake/mac/share/cmake-3.17/Modules/ExternalProject.cmake:1034
  (define_property):

    define_property command is not scriptable
...
```

So, looks like with this fix the workaround described in README is not required.

Mair, 2020-10-01:
yes. Another way would be
```
cmake_minimum_required(VERSION 3.5)
if (NOT DEFINED PROJECT_NAME)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
endif (NOT DEFINED PROJECT_NAME)
set(EXTRA_COMPONENT_DIRS components/lvgl_esp32_drivers components/lvgl_esp32_drivers/lvgl_touch components/lvgl_esp32_drivers/lvgl_tft)

if (NOT DEFINED PROJECT_NAME)
	project(lvgl-demo)
endif (NOT DEFINED PROJECT_NAME)
```