lvgl/lv_esp_idf #88: Set rotation of display on menuconfig

C47D, 2020-03-24:
The rotation of the display should be available for all supported displays.

C47D, 2020-04-14:
Supported in ILI9341 and ST7789 so far.

BFFonseca, 2020-05-04:
Hello.
Is there any way to set the rotation to 180º or 270º? This would be a good feature on menuconfig.

My display is upside down
[image]

Sorry if this isn't the right place to ask.

C47D, 2020-05-04:
@BFFonseca Hi, is that an ILI9341 based display?

BFFonseca, 2020-05-04:
> @BFFonseca Hi, is that an ILI9341 based display?

Yes! It is an ILI931 320x240 with an XPT2046.

I was able to rotate the screen by hardcoding on the ili_init_cmds[] structure's 0x36 command with a value of 0xE8, like so:
[image]

BFFonseca, 2020-05-04:
I am guessing this is not a very clean way to do it, but it was the only fix I could find.
I've started working on this lib only a couple days ago.
Would be nice if there was a way of getting the 4 possible rotations (0º, 90º, 180º and 270º) on the menuconfig.

C47D, 2020-05-04:
Don't worry, I just turned on my PC, I will let you know when I add the options to the menuconfig.

BFFonseca, 2020-05-05:
> Don't worry, I just turned on my PC, I will let you know when I add the options to the menuconfig.

Alright, looking forward to it.

By the way, I based myself on this library that I used in Arduino before: https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9341_Rotation.h

C47D, 2020-05-05:
Thanks for the link I will check it. I'm pushing my changes here: https://github.com/littlevgl/lv_port_esp32/tree/ili9341_rotation

Problem is the four orientations seems to work on the Wrover Kit v4.1 (it have an embedded display with the same ILI9341 controller), but only landscape mode works on the display with the separated ILI9341 controller (like yours).

C47D, 2020-05-06:
@BFFonseca Hi, i think i found the root issue, commit d6f6e18 on the `ili9341_rotation` branch seems to fix all 4 rotations, can you test that branch and let me know if it works for you?

I've tested it locally but is better if you can test it as well.

BFFonseca, 2020-05-06:
I can confirm it's fixed!
All 4 rotations and the touch work fine.
Touch coordinate are switched up, but that's to be expected.

Thank you! Excellent work.

Portrait:
[image]

Portrait Inverted:
[image]

Landscape:
[image]

Landscape Inverted:
[image]

Suggestion for future implementation: Auto-rotate the touch coordinated based on screen rotation.

C47D, 2020-05-06:
Thanks for testing it @BFFonseca :) , I think you can swap X and Y coordinates on the touch menu. I didn't tested it.

Sure, we could try to make the coordinate swap automagically, based on the rotation, let's leave that for another time.

BFFonseca, 2020-05-11:
@C47D no problem. Was a feature I needed, so thank you!

Yes, that is what I did, and it's not hard to do anyway. I would be a nice to have feature, but indeed not a priority.

Yes, you can use my pictures on the README :)

C47D, 2020-06-04:
All displays with rotation support are currently available in the menuconfig.

closed 2020-06-04 by C47D