lvgl/lv_esp_idf #90: Driver for 3.5inch RPi Display (MPI3501)

ndunello, 2020-03-31:
I developed and tested driver for the tft [MPI3501](http://www.lcdwiki.com/3.5inch_RPi_Display).
It's based on ILI9486 with custom circuit to convert 3-line SPI to 16-bit parallel. Touch controller is XPT2046.

Can I release it? If yes, how to proceed?

**ESP32 Chip version: ESP32D0WDQ6 (revision 1)**

**ESP-IDF version: 4.0**

**Development kit used: DevKitC**

**Development machine OS: Windows 10**

C47D, 2020-03-31:
Hi,

Great, I should add a contribution guide to add support for drivers, it's a easy but laborious process.
I think is best if we add the support for the ILI9486 controller and add the MPI3501 as predefined display, what do you think?

To do so:
1. In `components/lvgl_esp32_drivers/lvgl_tft/` copy `ili9341.h` and rename it after the display controller, so `ili9486.h`, replace the text `ILI9341` with `ILI9486`, this is the bare minimal, you can add register definitions, etc. it will end up something like this:
```c
#ifndef ILI9486_H
#define ILI9486_H

#ifdef __cplusplus
extern "C" {
#endif

/*********************
 *      INCLUDES
 *********************/
#include <stdbool.h>

#include "lvgl/lvgl.h"

/*********************
 *      DEFINES
 *********************/
#define DISP_BUF_SIZE (LV_HOR_RES_MAX * 40)
#define ILI9486_DC   CONFIG_LVGL_DISP_PIN_DC
#define ILI9486_RST  CONFIG_LVGL_DISP_PIN_RST
#define ILI9486_BCKL CONFIG_LVGL_DISP_PIN_BCKL

#define ILI9486_ENABLE_BACKLIGHT_CONTROL CONFIG_LVGL_ENABLE_BACKLIGHT_CONTROL

#if CONFIG_LVGL_BACKLIGHT_ACTIVE_LVL
  #define ILI9486_BCKL_ACTIVE_LVL 1
#else
  #define ILI9486_BCKL_ACTIVE_LVL 0
#endif

// if text/images are backwards, try setting this to 1
#define ILI9486_INVERT_DISPLAY CONFIG_LVGL_INVERT_DISPLAY

/**********************
 *      TYPEDEFS
 **********************/

/**********************
 * GLOBAL PROTOTYPES
 **********************/

void ili9486_init(void);
void ili9486_flush(lv_disp_drv_t * drv, const lv_area_t * area, lv_color_t * color_map);
void ili9486_enable_backlight(bool backlight);

/**********************
 *      MACROS
 **********************/

#ifdef __cplusplus
} /* extern "C" */
#endif

#endif /*ILI9486_H*/
```

2. Same for the c file, copy `ili9341.c` and rename it to `ili9486.c`, rename the functions `init`, `flush` and `enable_backlight` so they start with `ili9486`. Also rename all the `ILI9341` references to `ILI9486`.
Edit the functions so they match your init configuration sequence, flush function.

3. Open `disp_driver.h` and edit it so it end up like this:
```c
/*********************
[... 109 lines cut ...]
		bool "HX8357"
	    config LVGL_TFT_DISPLAY_CONTROLLER_ILI9486
		bool "ILI9486"
	endchoice
```

Then edit the LVGL_DISPLAY_WIDTH and LVGL_DISPLAY_HEIGHT with the size of the display, replace the XXX with the actual numbers:
```c
    config LVGL_DISPLAY_WIDTH
        int "TFT display width in pixels." if LVGL_PREDEFINED_DISPLAY_NONE
        default 240 if LVGL_PREDEFINED_DISPLAY_M5STACK
	default 480 if LVGL_PREDEFINED_DISPLAY_ERTFT0356 || LVGL_PREDEFINED_DISPLAY_ADA_FEATHERWING
        default XXX if LVGL_PREDEFINED_DISPLAY_MPI3501
	default 320

    config LVGL_DISPLAY_HEIGHT
        int "TFT display height in pixels." if LVGL_PREDEFINED_DISPLAY_NONE
        default 320 if LVGL_PREDEFINED_DISPLAY_M5STACK || LVGL_PREDEFINED_DISPLAY_ERTFT0356 || LVGL_PREDEFINED_DISPLAY_ADA_FEATHERWING
        default XXX if LVGL_PREDEFINED_DISPLAY_MPI3501
        default 240
```

If the display have a backlight enable pin edit the LVGL_ENABLE_BACKLIGHT_CONTROL config, add a field for the LVGL_PREDEFINED_DISPLAY_MPI3501 as we did with the width and heigth before.

I think that's everything you need to do, please let me know if you have any questions.

ndunello, 2020-03-31:
Hi @C47D,
thank you very much for the detailed explanation.
I had already made the changes you indicated.
This is the output of the **git status** command:
```
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   components/lv_examples/lv_ex_conf.h
        modified:   components/lvgl/lv_conf.h
        modified:   components/lvgl_esp32_drivers/lvgl_driver.c
        modified:   components/lvgl_esp32_drivers/lvgl_tft/Kconfig
        modified:   components/lvgl_esp32_drivers/lvgl_tft/disp_driver.c
        modified:   components/lvgl_esp32_drivers/lvgl_tft/disp_driver.h
        modified:   components/lvgl_esp32_drivers/lvgl_tft/disp_spi.c
        modified:   components/lvgl_esp32_drivers/lvgl_touch/xpt2046.c
        modified:   sdkconfig

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        .project
        components/lvgl_esp32_drivers/lvgl_tft/mpi3501.c
        components/lvgl_esp32_drivers/lvgl_tft/mpi3501.h

no changes added to commit (use "git add" and/or "git commit -a")

```
Also I created _mpi3501.c_ and _mpi3501.h_ in _components\lvgl_esp32_drivers\lvgl_tft_ directory.
So I wanted to know how to publish the changes for the community. I'm novice with the git :-(
Do I have to send you an email with the patch file?

Ciao

C47D, 2020-03-31:
I can help you to create a pull request if you want, or if you prefer you can send me your files, my mail is in my user main page.

To create a pull request:
you need to move those changes into a new branch, go to where youre working on, then create a new branch:

```
git checkout -b rpi_display
```

Then remove the sdkconfig from the staging area:
```
git reset sdkconfig
```

Add the new created files:
```
git add components/lvgl_esp32_drivers/lvgl_tft/mpi3501.c
git add components/lvgl_esp32_drivers/lvgl_tft/mpi3501.h
```

Do a commit:
```
git commit -m "Initial support for MPI3501"
```

Then push the commit to your fork:
```
git push origin rpi_display
```

Go to your fork and you will see a button to create a pull request with your additions into this repo, click it and that should be it.

ndunello, 2020-04-01:
I tried with the pull request, but the last command fails!
```
git status
On branch rpi_display
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   components/lv_examples/lv_ex_conf.h
        modified:   components/lvgl/lv_conf.h
        modified:   components/lvgl_esp32_drivers/lvgl_driver.c
        modified:   components/lvgl_esp32_drivers/lvgl_tft/Kconfig
        modified:   components/lvgl_esp32_drivers/lvgl_tft/disp_driver.c
        modified:   components/lvgl_esp32_drivers/lvgl_tft/disp_driver.h
        modified:   components/lvgl_esp32_drivers/lvgl_tft/disp_spi.c
        modified:   components/lvgl_esp32_drivers/lvgl_touch/xpt2046.c
        modified:   sdkconfig

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        .project
        add_mpi3501_support.patch

no changes added to commit (use "git add" and/or "git commit -a")

C:\projects\esp32\lv_port_esp32>git push origin rpi_display
remote: Permission to littlevgl/lv_port_esp32.git denied to ndunello.
fatal: unable to access 'https://github.com/littlevgl/lv_port_esp32.git/': The requested URL returned error: 403

```
I sent you an email with the patch of the modified files and the new files.
Thanks

C47D, 2020-04-01:
Hi,

Just checked my email and I don't see your email, did you sent the email to carlos.santiago.diaz@gmail.com?

embeddedt, 2020-04-01:
@ndunello You tried to push to https://github.com/littlevgl/lv_port_esp32 (which you don't have write access to). You need to fork that repository, clone the fork, and apply your changes there.

Here's the [blog post](https://blog.littlevgl.com/2018-12-06/contributing#how-can-i-send-fixes-and-improvements) we routinely link to for the main LittlevGL repository. The process is basically identical except you want to clone `lv_port_esp32` instead of `lvgl`.

closed 2020-04-01 by C47D

ndunello, 2020-04-02:
the pins that I used for the connection are the following (shared spi):
**display**

- MISO 19
- MOSI 23
- CLK 18
- CS 15
- DC 2
- RST 4

**touch**

- MISO 19
- MOSI 23
- CLK 18
- CS 22
- IRQ 25

[image]

the display maybe has some problems
[image]

I also attach the pinout of the display hoping it will be useful
[image]

C47D, 2020-04-02:
Thanks for the information @ndunello, I will update the README with it :)