lvgl/lv_esp_idf #92: Display flickers when ESP32 is in AP mode

eddeyonline, 2020-04-04:
I've noticed a consistent issue with the display flickering when the ESP32 is in AP or APSTA mode. I'm using a 240x320 2.4" TFT LCD from AliExpress with ILI9341 display driver and XPT2046 touch driver. The display works perfectly when wifi is not enabled or when the ESP32 is connected in STA mode. As soon as it is in AP or APSTA mode, without changing any other factors or connecting to the AP, the display has a constant flicker (maybe 20-30Hz just by looking at it). Any ideas on how to resolve this issue? Thanks!

C47D, 2020-04-04:
Could you upload some code to replicate it?

eddeyonline, 2020-04-05:
/* LVGL Example project
 * 
 * Basic project to test LVGL on ESP32 based projects.
 *
 * This example code is in the Public Domain (or CC0 licensed, at your option.)
 *
 * Unless required by applicable law or agreed to in writing, this
 * software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
 * CONDITIONS OF ANY KIND, either express or implied.
 */

// Change whether AP or STA mode with define at line 55, note flicker in AP mode
// Fill in SSID and password for STA mode at lines 85-86

#include <stdbool.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "freertos/event_groups.h"
#include "esp_event.h"

#include "esp_wifi.h"
#include "esp_log.h"
#include "esp_event_loop.h" //might not need this
#include "nvs_flash.h"
#include "esp_freertos_hooks.h"
#include "esp_system.h"

/* Littlevgl specific */
#include "lvgl/lvgl.h"

#include "disp_spi.h"
#include "disp_driver.h"
#include "tp_spi.h"
#include "touch_driver.h"

/*********************
 *      DEFINES
 *********************/

#define MODE_AP_STA 0  //if this is 0 ESP32 will boot in AP mode, 1 will boot in STA mode

// Detect the use of a shared SPI Bus and verify the user specified the same SPI bus for both touch and tft
#if (CONFIG_LVGL_TOUCH_CONTROLLER == 1 || CONFIG_LVGL_TOUCH_CONTROLLER == 3) && TP_SPI_MOSI == DISP_SPI_MOSI && TP_SPI_CLK == DISP_SPI_CLK
#if CONFIG_LVGL_TFT_DISPLAY_SPI_HSPI == 1
#define TFT_SPI_HOST HSPI_HOST
#else
#define TFT_SPI_HOST VSPI_HOST
#endif
 
#if CONFIG_LVGL_TOUCH_CONTROLLER_SPI_HSPI == 1
#define TOUCH_SPI_HOST HSPI_HOST
#else
#define TOUCH_SPI_HOST VSPI_HOST
#endif

#if TFT_SPI_HOST != TOUCH_SPI_HOST
#error You must specifiy the same SPI host for both display and input driver
#endif

#define SHARED_SPI_BUS
#endif

 #define EXAMPLE_ESP_MAXIMUM_RETRY  5
// FreeRTOS event group to signal when we are connected
static EventGroupHandle_t s_wifi_event_group;
/* The event group allows multiple bits for each event, but we only care about two events:
[... 300 lines cut ...]
  spi_bus_config_t buscfg = {
    .miso_io_num = TP_SPI_MISO,
    .mosi_io_num = DISP_SPI_MOSI,
    .sclk_io_num = DISP_SPI_CLK,
    .quadwp_io_num = -1,
    .quadhd_io_num = -1,
#if CONFIG_LVGL_TFT_DISPLAY_CONTROLLER == TFT_CONTROLLER_ILI9341
    .max_transfer_sz = DISP_BUF_SIZE * 2,
#elif CONFIG_LVGL_TFT_DISPLAY_CONTROLLER == TFT_CONTROLLER_ST7789
    .max_transfer_sz = DISP_BUF_SIZE * 2,
#elif CONFIG_LVGL_TFT_DISPLAY_CONTROLLER == TFT_CONTROLLER_ILI9488
    .max_transfer_sz = DISP_BUF_SIZE * 3,
#elif CONFIG_LVGL_TFT_DISPLAY_CONTROLLER == TFT_CONTROLLER_HX8357
    .max_transfer_sz = DISP_BUF_SIZE * 2
#endif
  };

  esp_err_t ret = spi_bus_initialize(TFT_SPI_HOST, &buscfg, 1);
  assert(ret == ESP_OK);

  /* SPI Devices */
  disp_spi_add_device(TFT_SPI_HOST);
  tp_spi_add_device(TOUCH_SPI_HOST);
}
#endif

eddeyonline, 2020-04-05:
That didn't format too well! I'll just email you Carlos.

C47D, 2020-04-05:
Could you create a repo with it?

eddeyonline, 2020-04-05:
I've never done that before, but I'll give it a go!

eddeyonline, 2020-04-10:
Repo:
https://github.com/eddeyonline/lv_port_esp32_flickering.git

C47D, 2020-04-11:
Thanks for the repo @eddeyonline, I will take a look at it.

KammutierSpule, 2020-04-12:
Is the flickering brightness change (a possible hardware power supply issue) or pixel filling flickering?

eddeyonline, 2020-04-27:
I don't think it's a power issue as changing power supply doesn't make a difference and it doesn't flicker when under load in STA mode.
I've just switched from using VSPI to HSPI to see if that made a difference but the issue is the same.
I've tried videoing the display in the two different modes to demonstrate the issue, but the flickering isn't apparent on the video. 
Is anyone else noting this issue at all?

closed 2020-05-25 by stale[bot]

showdown, 2024-11-07:
Was any resolution found to this? I'm seeing the same issue- a very high frequency flicker almost like a refresh when ESP32-S3 is in AP mode...