lvgl/lv_esp_idf #96: Menuconfig does not give HSPI/VSPI option for TFT

benton-anderson, 2020-04-14:
Latest update no longer includes option to set SPI port in Menuconfig for TFT controller. After saving Menuconfig, lvgl_spi_conf.h shows that both defines for `TFT_SPI_HOST` and `TOUCH_SPI_HOST` expand to `VSPI_HOST` 

When running the demo.h example I get the following printed to serial. 

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6876
load:0x40078000,len:14076
ho 0 tail 12 room 4
load:0x40080400,len:4304
entry 0x400806e8
I (71) boot: Chip Revision: 1
I (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v4.0 2nd stage bootloader
I (41) boot: compile time 00:27:27
I (52) boot: Enabling RNG early entropy source...
I (52) boot: SPI Speed      : 40MHz
I (52) boot: SPI Mode       : DIO
I (56) boot: SPI Flash Size : 2MB
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x42b44 (273220) map
I (212) esp_image: segment 1: paddr=0x00052b6c vaddr=0x3ffb0000 size=0x023a0 (  9120) load
I (216) esp_image: segment 2: paddr=0x00054f14 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/Users/ben/esp-idf-v4.0/components/freertos/xtensa_vectors.S:1778

I (219) esp_image: segment 3: paddr=0x0005531c vaddr=0x40080400 size=0x0a858 ( 43096) load
I (245) esp_image: segment 4: paddr=0x0005fb7c vaddr=0x00000000 size=0x00494 (  1172) 
I (246) esp_image: segment 5: paddr=0x00060018 vaddr=0x400d0018 size=0x2d710 (186128) map
0x400d0018: _stext at ??:?

I (325) boot: Loaded app from partition at offset 0x10000
I (325) boot: Disabling RNG early entropy source...
I (326) cpu_start: Pro cpu up.
I (329) cpu_start: Application information:
I (334) cpu_start: Project name:     esp_sunrise_alarm_clock
I (340) cpu_start: App version:      0075c49-dirty
I (346) cpu_start: Compile time:     Apr 14 2020 02:18:53
I (352) cpu_start: ELF file SHA256:  4f77b4f9d657a4a3...
I (358) cpu_start: ESP-IDF:          v4.0
I (363) cpu_start: Starting app cpu, entry point is 0x4008118c
0x4008118c: call_start_cpu1 at C:/Users/ben/esp-idf-v4.0/components/esp32/cpu_start.c:272

I (349) cpu_start: App cpu up.
I (373) heap_init: Initializing. RAM available for dynamic allocation:
I (380) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (386) heap_init: At 3FFC85F8 len 00017A08 (94 KiB): DRAM
I (392) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (399) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (405) heap_init: At 4008AC58 len 000153A8 (84 KiB): IRAM
I (411) cpu_start: Pro cpu start user code
I (430) spi_flash: detected chip: generic
I (430) spi_flash: flash io: dio
W (430) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (441) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (210) ILI9341: ILI9341 initialization.
E (410) spi: SPI3 already claimed by spi master.
E (410) spi_master: spi_bus_initialize(231): host already in use
assertion "ret==ESP_OK" failed: file "../components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_touch/tp_spi.c", line 92, function: tp_spi_init
abort() was called at PC 0x400ef72f on core 1
0x400ef72f: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

ELF file SHA256: 4f77b4f9d657a4a3d5aab77d92d449c0c718810cb575cdb86541e3403626f323

Backtrace: 0x4008512d:0x3ffcdd90 0x400854a5:0x3ffcddb0 0x400ef72f:0x3ffcddd0 0x400d5189:0x3ffcde00 0x400d50e1:0x3ffcde40 0x400d4344:0x3ffcde60 0x400d4261:0x3ffcde80 0x400882a5:0x3ffcdf00
0x4008512d: invoke_abort at C:/Users/ben/esp-idf-v4.0/components/esp32/panic.c:155

0x400854a5: abort at C:/Users/ben/esp-idf-v4.0/components/esp32/panic.c:172

0x400ef72f: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x400d5189: tp_spi_init at C:\Users\ben\esp_sunrise_alarm_clock\build/../components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_touch/tp_spi.c:92 (discriminator 1)

0x400d50e1: touch_driver_init at c:\users\ben\esp_sunrise_alarm_clock\build/../components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_touch/touch_driver.c:13

0x400d4344: lvgl_driver_init at C:\Users\ben\esp_sunrise_alarm_clock\build/../components/lv_port_esp32/components/lvgl_esp32_drivers/lvgl_driver.c:63

0x400d4261: guiTask at C:\Users\ben\esp_sunrise_alarm_clock\build/../main/main.c:69

0x400882a5: vPortTaskWrapper at C:/Users/ben/esp-idf-v4.0/components/freertos/port.c:143
```

The TFT screen flickers white and resets within 1 second. The above message is printed to serial every time.

**ESP32 Chip version** DevKit Rev1

**ESP-IDF version** 4.0

**Development kit used** Eclipse ESP IDF

**Development machine OS** Windows

benton-anderson, 2020-04-14:
Can confirm that the `main.c` example works after modifying `disp_spi.c` to replace `TFT_SPI_HOST` with `1` at the following lines:

```
    //Initialize the SPI bus
    ret=spi_bus_initialize(TFT_SPI_HOST, &buscfg, 1);
    assert(ret==ESP_OK);

    //Attach the LCD to the SPI bus
    disp_spi_add_device(TFT_SPI_HOST);
```

C47D, 2020-04-14:
Thanks for the report, can you try using this branch? https://github.com/littlevgl/lv_port_esp32/tree/restore_spi_bus_option
The menuconfig now shows the SPI Bus selection option, but I have no hardware to test it until later today.

benton-anderson, 2020-04-14:
Yes, Menuconfig works and the demo works too. Thank you.

closed 2020-04-14 by benton-anderson

C47D, 2020-04-14:
No problem, I will update the master branch with the fix. Thanks for testing it.