lvgl/lv_esp_idf #101: Port to LVGL v7.0

C47D, 2020-04-21:
List of necessary changes:

- [ ] Update lvgl submodule.
- [ ] Update lvgl example submodule.
- [ ] Update lvgl configuration file.

kisvegabor, 2020-04-21:
>  Update lvgl submodule.

It's in `dev-7.0` branch

> Update lvgl example submodule.

It's in `rework-7` branch

> Update lvgl configuration file.

It'd be best to create a new one based on `lv_conf_templ.h` in `dev-7.0`.

Could you make a build test to see if there are any serious issues?

C47D, 2020-04-22:
I will do the tests here: https://github.com/C47D/lv_port_esp32_v7
Once we get it working I will update this repo

C47D, 2020-04-23:
First teaser, still some work to do i think

[image]

kisvegabor, 2020-04-23:
What is the size and resolution of the display and LV_DPI?

C47D, 2020-04-23:
Size of the screen is 320*240, LV_DPI is 100.

[image]

kisvegabor, 2020-04-23:
The demo recognizes if it's a small, medium, large or extra large display and sets the layouts accordingly. This display should have been recognized as small. 
What is the size of the display (in inches). We can calculate the real DPI from it.

Besides, please modify `LV_DISP_SMALL_LIMIT` to `25` in `lv_conf.h`.

C47D, 2020-04-23:
The LCD size is 3.2" (inches), ok I will modify `LV_DISP_SMALL_LIMIT` to 25 and upload a pic.

kisvegabor, 2020-04-23:
So the actual DPI is: `sqrt(320^2 + 240^2) / 3.2 = 125.
And width of the display is 320/125 = 2.56".
 
If you set to `LV_DISP_SMALL_LIMIT` to 25 lvgl will consider > 2.5" display as medium-sized and create this ugly layout.  So please set it to `30` (3.0") instead of `25`.

C47D, 2020-04-23:
Much better

[image]

kisvegabor, 2020-04-23:
Awesome, thank you!
I'll recalculate the default size limits.

You can enable `LV_USE_PERF_MONITOR` in `lv_conf.h` to see the current FPS and CPU usage.

C47D, 2020-04-23:
I'm getting 33FPS, with CPU from 18% to 28% and peaks of 34%.

embeddedt, 2020-04-23:
@C47D I'm curious; what FPS do you get at the moment when you scroll past the gauges in the Values tab? I'd like to compare with what I see on STM32F7 (I get 14-18 FPS while scrolling; it increases after that).

C47D, 2020-04-23:
Hi @embeddedt, this particular display doesn't have a touch controller, I will try to setup another display to measure the FPS values you've requested.

kisvegabor, 2020-04-24:
@C47D I've modified the demo in [test/no-tp](https://github.com/littlevgl/lv_examples/tree/test/no-tp) branch. It changes between the tabs automatically and shows the gauges on the second tab.

C47D, 2020-04-24:
Hi, I took a video of the demo using the test/no-tp branch.

https://youtu.be/BJv-vr03RsM

kisvegabor, 2020-04-24:
Thank you!

There is still huge FPS drop when the gauge is fully redrawn. At least refreshing the needle only is not that bad. 

I have a few questions:
- What is the size of the display buffer?
- Do you use 2 buffers with DMA in `flush_cb`?
- Have you enabled `-Os` or `-O3` optimization?

C47D, 2020-04-24:
Hi, I'm not at home right now I will update my reply with the data you requested.

I had an issue with the demo, the demo stopped working after some minutes and the screen is stuck, I have tried to replicate the issue but it stopped at different times. I will try to add the log functions to lvgl so I can know when it stops.

embeddedt, 2020-04-24:
> the demo stopped working after some minutes and the screen is stuck

I also saw this happen once on my STM32F7, but I didn't have time to investigate further. It was a few days ago.

C47D, 2020-04-24:
Added some printf debugging points to see where the display is getting stuck and I can't replicate the issue, the demo has been running for almost an hour, but noticed that the `lv_tick_task` is still running.

There's also a new issue on the lvgl repo which uses the same dev board as me and doesn't report the issue [Dev-7.0 performance experiments](https://github.com/littlevgl/lvgl/issues/1471)

kisvegabor, 2020-04-24:
I also saw the freeze but I thought it's a bug in my hacky display driver. I've already debuged that it stops on a `while(vdb->flushing);` I still don't know if `lv_disp_flush_ready` is not called or it's called  but as it is called from an interrupt (for all of us) it might mess up things somehow. I mean some assembly level you-never-find-it bug. 

I suspect the latest because `while(vdb->flushing);` reads a  bit field which can be quite complicated on assembly level.

kisvegabor, 2020-04-24:
I suggest running the `lv_demo_stress()` which makes much more drawing.

embeddedt, 2020-04-24:
I call `lv_disp_flush_ready` directly from the `flush_cb` function, so it can't be entirely an interrupt-related issue.

kisvegabor, 2020-04-24:
> I call lv_disp_flush_ready directly from the flush_cb function, so it can't be entirely an interrupt-related issue.

Ah, good to know. I thought you are using the DMA based `flush_cb`.

C47D, 2020-04-24:
I call `lv_disp_flush_ready` from a callback when the SPI finished transferring the data via DMA.

C47D, 2020-04-24:
@kisvegabor Do you know if it's possible to get a backtrace when the application is stuck? I'm searching some tutorials to debug the esp32 chip with gdb.

embeddedt, 2020-04-24:
If you have debug symbols available and can attach to the device with GDB after it's already running, you should be able to interrupt execution, type `bt`, and get a decently usable backtrace.

kisvegabor, 2020-04-24:
It seems @embeddedt knows it much better than me :slightly_frowning_face: 

---

I added some debug code and will run the stress demo all night, and hopefully it'll freeze.

kisvegabor, 2020-04-24:
I think I found the issue but the picture is not perfect yet.

[These](https://github.com/littlevgl/lvgl/blob/4502d7a879c6bd0050c650e525eb3e96ff957a4b/src/lv_hal/lv_hal_disp.h#L55-L58) are the interesting parts of `lv_disp_buf_t`. So there are 4 bits next to each other. When going to the next part to refresh lvgl might write the last 3 fields. So what I suspect is:
1. When e.g. `last_area` is set, first the whole bitfield is read to a register
2. An interrupt comes and set `flushing = 0`
3. In the register read in 1) the `last_area` is set but in the register,`flushing` is still 1 because it was cleared in the "real" variable.
4. The register is written back and it overwrites the `flushing` bit.

So it's a typical Read-Modify-Write issue.
It all makes sense but @embeddedt said he doesn't use interrupt in the `flush_cb`.
So @embeddedt, are you sure you weren't using an interrupt based driver when it froze for you?

I've pushed a [trivial fix](https://github.com/littlevgl/lvgl/blob/15e064f4af5c68bf5f9880da23a4de0794288cb4/src/lv_hal/lv_hal_disp.h#L55-L58). Let's see if it helps.

C47D, 2020-04-24:
Thanks for the explanation @kisvegabor, I've set the tools to be able to debug the demo-application (I'm still using the demo-widgets app), I'm running it with gdb so I can backtrace when the application gets stuck.

The "problem" I see is that I we don't know how much time it will take to the application to stall, so how much time should we run the application with the fix to be sure it worked as expected?

embeddedt, 2020-04-24:
> are you sure you weren't using an interrupt based driver when it froze for you?

100% sure. I don't have access to interrupts in the context that LittlevGL runs in. My setup is a bit unique so it's quite possible that the issue isn't within LittlevGL (although I didn't experience this with 6.1 or earlier snapshots of 7.0).

> how much time should we run the application with the fix to be sure it worked as expected?

We have several weeks before release, so I'm pretty sure one of us will run into the issue if it's still present. I'd say let it run for a few hours and if nothing goes wrong, we can consider it's fixed for now.

C47D, 2020-04-25:
@kisvegabor I've fixed an error on the lv_examples repo that I found when trying to run the stress demo, I sent a pull request to that repo.

Is this the expected behavior of the demo? https://youtu.be/7goD6lRqTLc

embeddedt, 2020-04-25:
I think that's what it's supposed to look like - the idea is that it randomly creates and moves a bunch of objects around to try and trigger weird bugs like this one.

I've had `lv_demo_widgets` running for a few hours now - it hasn't crashed. I think I can safely say the freeze is gone for me.

C47D, 2020-04-25:
I also haven't hit the bug again, I will leave the demo running overnight.

You guys know lvgl better than I do, is there any particular reason why `flushing` and `flushing_last` are ints instead of uint32_t?

kisvegabor, 2020-04-25:
> @kisvegabor I've fixed an error on the lv_examples repo that I found when trying to run the stress demo, I sent a pull request to that repo.
Is this the expected behavior of the demo? https://youtu.be/7goD6lRqTLc

Thank you for the PR. Yes, it should look something like this. You can increase `TIME_STEP` in `lv_demo_stress.c` (e.g. to 200) to see more.

It was running for me for more than 8 hours. So it really seems to be solved. :tada: 

Okay, let's turn back to the original topic: v7 on ESP.
@C47D seemingly it's running well on ESP. It could be faster though... On STM32F7, in general, I measured higher FPS with v7 compared to v6, but on ESP for a 320x240 TFT, I'd expect higher FPS.
Could you answer [these questions](https://github.com/littlevgl/lv_port_esp32/issues/101#issuecomment-619056551), please, to exclude some possibilities?

barbiani, 2020-04-25:
@C47D 
> I do have debug symbols available, but it's a pain to setup the ESP32 tools on Windows, I was using the WSL but can't run openocd there, so i have to setup the toolchain in Windows directly :/.

actually you  [can](https://www.esp32.com/viewtopic.php?f=13&t=9160&p=38216#p38216) or use [https://visualgdb.com/](https://visualgdb.com/)

C47D, 2020-04-25:
Hi @kisvegabor,

- What is the size of the display buffer?
For this particular display is `LV_HOR_RES_MAX * 40`, so 320 * 40 bytes, 12.5KBytes.

- Do you use 2 buffers with DMA in flush_cb?
This is the `flush_cb` of the display driver available on my dev kit:
https://github.com/littlevgl/lv_port_esp32/blob/d124fe22a99580c47b7e45f449faf028390d4353/components/lvgl_esp32_drivers/lvgl_tft/ili9341.c#L150-L177

But we configure with 2 buffers the display buffer:
https://github.com/littlevgl/lv_port_esp32/blob/d124fe22a99580c47b7e45f449faf028390d4353/main/main.c#L71-L74

- Have you enabled -Os or -O3 optimization?

No, the default project is being compiled without optimizations and debug symbol generation enabled, here are the possible optimization levels available in esp-idf.
[image]

- What is the speed of the SPI?

The speed of the SPI is 40MHz for this particular driver.

embeddedt, 2020-04-25:
Can you try it with `-Og` and see what happens? That should already be an improvement over no optimization.

C47D, 2020-04-25:
@embeddedt the project is being compiled with - 0g, I can try the other options tho if you want me to.

kisvegabor, 2020-04-25:
@C47D 
Just a minor correction: one buffer has 320*40*2 byte = 25 kB (because there is 2 bytes/pixel)

Could you try `-O2` and `-Os` too?

kisvegabor, 2020-04-25:
I usually don't use `-Og` so tested how it compares to others on my STM32F7 dev board:
```
-O0: 7 FPS
-Og: 18 FPS
-O2: 25 FPS
-Os: 24 FPS
```
So hopefully you also will see ~50% performance boost.

barbiani, 2020-04-25:
@C47D 

Did you do anything do add the -O2 setting? Mine shows only debug amd release.

C47D, 2020-04-25:
@barbiani I'm using the master branch of the esp-idf, maybe is because of that.

C47D, 2020-04-25:
@kisvegabor this is the demo stress compiled with -02

https://youtu.be/lTgXsJjYHk8

C47D, 2020-04-25:
> actually you can or use https://visualgdb.com/

Thanks for the information @barbiani, I ended up running openocd on the Windows prompt and gdb on WSL, but I had to change the drivers for the FTDI chip on windows to be able to connect openocd to it.
I debugged the application and it ran for about 5 hours with no issues, so i think the patch @kisvegabor did worked.

barbiani, 2020-04-25:
While writing a touch driver I have found that lvgl crashes with touches
outside of the screen. Meaning coordinates bigger than x and y resolutions.

kisvegabor, 2020-04-26:
> I debugged the application and it ran for about 5 hours with no issues, so i think the patch @kisvegabor did worked.

Awesome! :slightly_smiling_face: 

> While writing a touch driver I have found that lvgl crashes with touches
outside of the screen. Meaning coordinates bigger than x and y resolutions.

I'll check it!

kisvegabor, 2020-04-26:
> >While writing a touch driver I have found that lvgl crashes with touches
outside of the screen. Meaning coordinates bigger than x and y resolutions.

> I'll check it!

Should work now.

kisvegabor, 2020-04-26:
@C47D 
> this is the demo stress compiled with -02

It seems to me it's quite the same FPS as `-Og`. 

Sending 320 * 240 * 16bit with 40 MHz takes 30 ms (33 FPS) so it should not be a limiting factor (as it happens in parallel with rendering). However, a 200 MHz MCU should be much faster with a 320x240 TFT. It'd be awesome to see how does it look on an oscilloscope to send an empty screen to the driver (CLK would be enough). I know it needs more effort to do, so I'm just writing it, in case you or @barbiani  has time, interest an oscilloscope to measure it. :slightly_smiling_face:

C47D, 2020-04-26:
@kisvegabor, sorry if i say something silly but, aren't we limited by the spi clock and not the cpu clock?

I don't have an oscilloscope at home to test it :/, I'm going to the office until next Wednesday so i can test it until then, seems like i need a better logic analyzer :)

Regards

kisvegabor, 2020-04-26:
> sorry if i say something silly but, aren't we limited by the spi clock and not the cpu clock?

From the quick calculation above, we can see that it takes only 30 ms the send a full screen image. So, it limits the FPS to 33. If data sending to the display happens in the background (using DMA) the MCU is free to deal with rendering. If rendering takes < 30 ms we are still limited by the SPI clock speed, but the limit is still at 33 FPS. So is we see 16 FPS it means rendering took 60 ms. Which is way too much for a so small display and a 160 (or 240?) MHz MCU.

Could you try the widgets demo again with `-O2`? The stress demo could be misleading because it's not a realistic load.

C47D, 2020-04-26:
Thanks for the explanation @kisvegabor, I'm compiling the widgets demo with -O2, will upload the video asap.

C47D, 2020-04-26:
Here's the video for demo widgets compiled with -O2 https://youtu.be/6FWhe9Ovd6k

kisvegabor, 2020-04-27:
@C47D Thank you for the video. It still looks quite the same FPS as with `-Og`. :( 

I'm looking forward to seeing the oscilloscope measurement on Wednesday! :)

kisvegabor, 2020-04-27:
@C47D 
I've added some minor fixes to `lv_demo_bechmark`.
Could you try it and share a photo/video about the summary screen at the end?

embeddedt, 2020-04-27:
I ran `lv_demo_benchmark` on STM32F746 (without the GPU). This looks like a really great tool. Here's the results I'm seeing. For brevity I'm only including the ones that either stuttered visually or have poor FPS numbers.

* Small/medium text - 18 FPS
* Shadows -  ~15 FPS when small, ~11 FPS when an offset is added, <7 FPS when they're large. They seem to be very taxing.
* Image (A)RGB rotate/zoom anti-aliased - ~11 FPS (regardless of opacity or alpha)
* Medium/large *compressed* text - 8/4 FPS, respectively
* Substr. shadow - 9 FPS
* Substr. text - 15 FPS

embeddedt, 2020-04-27:
I should have mentioned before that all of my tests thus far without GPU are done with a custom driver that does direct copies to the framebuffer in `flush_cb`, and not with the official TFT driver, which uses standard DMA in `flush_cb`. I only use the official TFT driver to test with the GPU.

Results with the official TFT driver (GPU enabled):

* "Opa speed" increased from 79% to 91%.
* All uncompressed text now renders at ~11 FPS regardless of size.
* Circle borders now render at 16 FPS (they weren't on the "Slow but common" list before).
* Circles without opacity render at 10 FPS, but with opacity they drop to 13 FPS.
* Shadows now render at 8 FPS when small, 5 FPS when an offset is added, and 2 FPS when they're large. :confused: It seems that the GPU does a worse job with this (2-3x the speed without GPU).
* Image RGB renders at a surprising 173 FPS. Is this possible?
* Rotated images render at 15-20 FPS, adding anti-aliasing cuts that in half.
* Compressed text renders at 8 FPS when small, 6 FPS when medium, and 2 FPS when large.
* Substr. shadow - 4 FPS

kisvegabor, 2020-04-27:
@embeddedt 
It'd be interesting to measure with `speed/gpu` branch and enable `LV_USE_GPU_STM32_DMA2D` to use the builtin GPU support.

I've added `LV_SHADOW_CACHE_SIZE` to `lv_conf.h` to speed up shadow drawing. You should set it to 10000.

In general, it's strange that I see 2-3 times better FPS on STM32F769 which is quite the same MCU.
- Still disabled caching? (It'd explain the slowness)
- What is the optimization level? (-O3 for me)
- What is the display buffer size?
- Display buffers are in internal RAM? 
- Are you using the latest lvgl and lv_examples versions?

@C47D 
- The ESP dev board there is no external RAM, right?
- I saw that there is some kind of flash caching in ESP but still don't know what it is really good for. Do you know anything about it?
- I read this [doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/general-notes.html#memory-layout) and it seems it can cause some speed up if we'd place the performance-critical functions to IRAM.

embeddedt, 2020-04-27:
> It'd be interesting to measure with `speed/gpu` branch and enable `LV_USE_GPU_STM32_DMA2D` to use the builtin GPU support.

Will do.

> Still disabled caching? (It'd explain the slowness)

Yes; that's probably why it's so slow (although I would hope that it can perform reasonably even without the cache).

> What is the optimization level?

I'm using `-Og` at the moment. 

> Are you using the latest lvgl and lv_examples versions?

Always. :slightly_smiling_face: 

> What is the display buffer size? Display buffers are in internal RAM?

Thanks for checking that. I had forgotten to increase the size in the demo project, so it's currently at about 1/27th of the display size. We're probably losing a lot of performance to all the style calculations.

I'll increase the buffer size and optimization level, change `LV_SHADOW_CACHE_SIZE`, and report back with the results from `speed/gpu`.

C47D, 2020-04-27:
@kisvegabor @embeddedt how often should I call `lv_tick_inc`? Right now it's called every 10ms.

kisvegabor, 2020-04-27:
In general, it shouldn't matter too much, however it's better to keep below 5 ms for higher precision.

In this case, when lvgl needs to measure 10-100 ms (the rendering time) the 10 ms tick resolution can cause some measurement error. E.g. if the real rendering time is 45 ms (22.2  FPS), lvgl can measure i as 40 ms (25 FPS) or 50 ms (20 FPS).

[65 more replies]

closed 2020-06-04 by C47D