lvgl/lv_esp_idf #103: ILI9488 build error

persello, 2020-04-25:
**Describe the issue**
Error when compiling and ILI9488 is selected.

**Code to reproduce the issue**
Clone the repository, enter the menuconfig, set ILI9488 as the display driver and 480x320 resolution. Set the bus to HSPI.

**Expected Results**
The example compiles and runs.

**Actual Results**
idf.py flash fails.

NOTE: With the same settings, but ILI9341 selected instead of ILI9488, the build succeeds.

**ESP32 Chip version**
ESP-WROOM-32

**ESP-IDF version**
4.0

**Development machine OS**
Fedora 32 and macOS Catalina

**Compilation warnings/errors (if available)**

```
[10/72] Building C object esp-idf/lvgl_tft/CMakeFiles/__idf_lvgl_tft.dir/sh1107.c.obj
FAILED: esp-idf/lvgl_tft/CMakeFiles/__idf_lvgl_tft.dir/sh1107.c.obj 
/home/riccardo/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DLV_CONF_INCLUDE_SIMPLE=1 -Iconfig -I../components/lvgl_esp32_drivers/lvgl_tft -I/home/riccardo/esp/esp-idf/components/newlib/platform_include -I/home/riccardo/esp/esp-idf/components/freertos/include -I/home/riccardo/esp/esp-idf/components/freertos/xtensa/include -I/home/riccardo/esp/esp-idf/components/heap/include -I/home/riccardo/esp/esp-idf/components/log/include -I/home/riccardo/esp/esp-idf/components/lwip/include/apps -I/home/riccardo/esp/esp-idf/components/lwip/include/apps/sntp -I/home/riccardo/esp/esp-idf/components/lwip/lwip/src/include -I/home/riccardo/esp/esp-idf/components/lwip/port/esp32/include -I/home/riccardo/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/riccardo/esp/esp-idf/components/soc/src/esp32/. -I/home/riccardo/esp/esp-idf/components/soc/src/esp32/include -I/home/riccardo/esp/esp-idf/components/soc/include -I/home/riccardo/esp/esp-idf/components/esp_rom/include -I/home/riccardo/esp/esp-idf/components/esp_common/include -I/home/riccardo/esp/esp-idf/components/esp_system/include -I/home/riccardo/esp/esp-idf/components/xtensa/include -I/home/riccardo/esp/esp-idf/components/xtensa/esp32/include -I/home/riccardo/esp/esp-idf/components/esp32/include -I/home/riccardo/esp/esp-idf/components/driver/include -I/home/riccardo/esp/esp-idf/components/driver/esp32/include -I/home/riccardo/esp/esp-idf/components/esp_ringbuf/include -I/home/riccardo/esp/esp-idf/components/efuse/include -I/home/riccardo/esp/esp-idf/components/efuse/esp32/include -I/home/riccardo/esp/esp-idf/components/espcoredump/include -I/home/riccardo/esp/esp-idf/components/esp_timer/include -I/home/riccardo/esp/esp-idf/components/soc/soc/esp32/. -I/home/riccardo/esp/esp-idf/components/soc/soc/esp32/include -I/home/riccardo/esp/esp-idf/components/soc/soc/esp32/../include -I/home/riccardo/esp/esp-idf/components/vfs/include -I/home/riccardo/esp/esp-idf/components/esp_wifi/include -I/home/riccardo/esp/esp-idf/components/esp_wifi/esp32/include -I/home/riccardo/esp/esp-idf/components/esp_event/include -I/home/riccardo/esp/esp-idf/components/esp_netif/include -I/home/riccardo/esp/esp-idf/components/esp_eth/include -I/home/riccardo/esp/esp-idf/components/tcpip_adapter/include -I/home/riccardo/esp/esp-idf/components/app_trace/include -I../components/lvgl -I../components/lvgl/lvgl -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2-dev-1206-g741960d5c\" -DESP_PLATFORM -MD -MT esp-idf/lvgl_tft/CMakeFiles/__idf_lvgl_tft.dir/sh1107.c.obj -MF esp-idf/lvgl_tft/CMakeFiles/__idf_lvgl_tft.dir/sh1107.c.obj.d -o esp-idf/lvgl_tft/CMakeFiles/__idf_lvgl_tft.dir/sh1107.c.obj   -c ../components/lvgl_esp32_drivers/lvgl_tft/sh1107.c
../components/lvgl_esp32_drivers/lvgl_tft/sh1107.c: In function 'sh1107_set_px_cb':
../components/lvgl_esp32_drivers/lvgl_tft/sh1107.c:47:27: error: 'byte_index' may be used uninitialized in this function [-Werror=maybe-uninitialized]
 #define BIT_SET(a,b) ((a) |= (1U<<(b)))
                           ^~
../components/lvgl_esp32_drivers/lvgl_tft/sh1107.c:121:14: note: 'byte_index' was declared here
     uint16_t byte_index;
              ^~~~~~~~~~
../components/lvgl_esp32_drivers/lvgl_tft/sh1107.c:47:33: error: 'bit_index' may be used uninitialized in this function [-Werror=maybe-uninitialized]
 #define BIT_SET(a,b) ((a) |= (1U<<(b)))
                                 ^~
../components/lvgl_esp32_drivers/lvgl_tft/sh1107.c:122:14: note: 'bit_index' was declared here
     uint8_t  bit_index;
              ^~~~~~~~~
cc1: some warnings being treated as errors
```

I attach a log of the build output after running `idf.py clean`.
[log.log](https://github.com/littlevgl/lv_port_esp32/files/4534346/log.log)

C47D, 2020-04-26:
Thanks for the report @persello I will take a look at it.

persello, 2020-04-26:
It works, thanks!

PR #104 (2020-04-26): Fix ili9488 orientation
Fixes #103
closed 2020-04-26 by persello