lvgl/lv_esp_idf #105: cannot build disp_driver.c

KammutierSpule, 2020-04-26:
**Describe the issue**

Cannot build

**Compilation warnings/errors (if available)**
../components/lvgl_esp32_drivers/lvgl_tft/disp_driver.c:20:5: error: implicit declaration of function 'ili9341_init';

It looks it is adding this file and it shouldn't? because disp_driver.c is inline included in lvgl_driver.c

C47D, 2020-04-26:
Hi,

Are you selecting a predefined display configuration on the menuconfig or selecting every option by yourself? I'm working with the Wrover Kit v4.1 preconfiguration (it have the ILI9341 display controller) and i can't reproduce the issue.

KammutierSpule, 2020-04-26:
I'm assigning the PINs myself on the "Display Pin Assignments"
The predefined options are set to "None"
but I should note that CONFIG_LVGL_TFT_DISPLAY_CONTROLLER_ILI9341 is set
so this issues is because that .c file is included and there is no include to this define.
but I think thats ok because it should be excluded from the build list? ..

KammutierSpule, 2020-04-26:
I found the issue, It was my fault! I deleted by mistake the includes that are need to this esp32 port in the config.h

closed 2020-04-26 by KammutierSpule

C47D, 2020-04-26:
Don't worry, happy to know you solved it.