lvgl/lv_esp_idf #106: cannot build LV_HOR_RES_MAX and LV_VER_RES_MAX must be greater than 0

KammutierSpule, 2020-04-26:
**Describe the issue**

I'm trying to build V7 it used to work but with some of new updates it now cannot build and gives this error

**Code to reproduce the issue**
Using dev-7.0 of lvgl

**Compilation warnings/errors (if available)**

`../components/lvgl/lvgl/src/lv_widgets/../lv_core/lv_obj.h:39:2: error: #error "LittlevGL: LV_HOR_RES_MAX and LV_VER_RES_MAX must be greater than 0"
`

C47D, 2020-04-26:
Hi,

I also had that issue when trying to run lvgl v7, can you try using this repo? https://github.com/C47D/lv_port_esp32_v7 , once we figure out some issues I will import the changes to here.

KammutierSpule, 2020-04-26:
I found the issue, It was my fault! I deleted by mistake the includes that are need to this esp32 port in the config.h

closed 2020-04-26 by KammutierSpule

C47D, 2020-04-26:
No problem, please let me know if you have any more issues.

KammutierSpule, 2020-04-26:
> can you try using this repo? https://github.com/C47D/lv_port_esp32_v7
Thanks I'm now using this repo!