lvgl/lv_esp_idf #111: blank display with ILI9486

riks5, 2020-05-04:
If you try to connect a controller ILI9486 through a none predefined wiring: build and flash runs fine. After reboot the display keeps empty, touchscreen functions well.

In the preview of the sdkconfig one can see that the configuration sets 2 display controllers at the same time.

Is this a bug or is such a setup not possible at the moment?

C47D, 2020-05-04:
Sounds like a bug, can you append the sdkconfig.h file generated by idf? Or an screenshot of the two enabled display controllers?

C47D, 2020-05-04:
Never mind I think I found the issue, 42d51f9 must fix it, please update your repo, do `idf.py fullclean` and then configure the project again with the ILI9486 display controller, let me know if it fixes it.

Thanks for your report.

riks5, 2020-05-05:
Great progress, now display shows demo content!
Not vertically but horizontally flipped, see pic.
Changing "invert display" to off in config, sets the screen in portrait? and "not synced"

I attached sdkconfig.h and sdkconfig.old
Same wiring works fine in other project/driver.

Thanks for your help!

[sdkconfig.txt](https://github.com/littlevgl/lv_port_esp32/files/4580446/sdkconfig.txt)
[sdkconfigold.txt](https://github.com/littlevgl/lv_port_esp32/files/4580448/sdkconfigold.txt)

[image]
[image]

C47D, 2020-05-05:
Hi,

Thanks for the pics and files, yeah, last night I was struggling with the Ili9341 controller having similar issues.

I don't have your exact driver, I will need you to test some changes for me, can you?. I will let you know when I upload them.

riks5, 2020-05-05:
Okay, I understand.
Yes I can do some testing for you, but hey... I'm not a pro.
I would like to use the littlevgl-driver in the future.

C47D, 2020-05-05:
> Okay, I understand.
> Yes I can do some testing for you, but hey... I'm not a pro.
> I would like to use the littlevgl-driver in the future.

Don't worry, is easy. let's try something fast (until i get out of my work). Keep invert display in off.

In file `components/lvgl_esp32_drivers/lvgl_tft/ili9486.c` can you edit the line:
https://github.com/littlevgl/lv_port_esp32/blob/42d51f924bff20904a0e5fdeaf9fcf3f772c72f6/components/lvgl_esp32_drivers/lvgl_tft/ili9486.c#L61

To:
```c
{0x36, {0x08}, 1}
```
Then compile and flash the program.

After that change the same line to this:
```c
{0x36, {0x68}, 1}
```
Then compile and flash the program.

After that change the same line to this:
```c
{0x36, {0xC8}, 1}
```
Then compile and flash the program.

After that change the same line to this:
```c
{0x36, {0xA8}, 1}
```
Then compile and flash the program.

Check if that values work, if so, please take pics of it.

Register `0x36` is the register that handles the orientation of the displayed data.

riks5, 2020-05-05:
Sorry, cann't send any pic with goog screen.
Screen is never blank, content rotates but is not okay.

Only with 0x48 the txt on screen is readable but in portrait as config indicates landscape.

Maybe important: warnings during build.
[image]

C47D, 2020-05-05:
I will fix that warning later today. So 0x48 works as portrait, noted that.

Can you try this new values? First `0x28`, then `0x88` and `0xE8`.

Sorry for the inconvenience, but i don't have that display to try myself :/

riks5, 2020-05-05:
0xE8.....bingo!!! 
See pics, well done.

[image]
[image]
[image]

C47D, 2020-05-05:
Great 😄 , did you had the chance to try the other values?

riks5, 2020-05-05:
Yes, I tried all values.
Only one works fine: text and view 100% okay.
Attention/correction: with 0x48 only text was readable, part of screen was gray in portrait.

C47D, 2020-05-05:
Fck, I need to fully understand how the displays are being updated, just for reference I got those values from here: https://github.com/schreibfaul1/ESP32-TFT-Library-ILI9486/blob/master/src/tft.cpp

riks5, 2020-05-05:
Sorry I cann't help you with that. 
I found and read the manual, full of clear explications but concepts of digital media way too complicated for me.
I'm just a newbie in the firmware world.

C47D, 2020-05-05:
Don't worry, we will figure it out, thanks for the help and patience.

riks5, 2020-05-06:
Good luck with that.
Many thanks for your help and fixing problem!

C47D, 2020-05-06:
@riks5 I found a bug last night that made screens see like your first picture, the one that is incomplete and I think I fix it.

I will add the necessary code to make your display able yo rotate later today, can you help me out and test it? I can upload the instructions later today.

riks5, 2020-05-06:
I saw your reopening just now.
Good improvement!
All layouts work just fine.
I noticed with every build a warning, see pics of 2 versions.

[image]
[image]

C47D, 2020-05-06:
Hi, thanks for taking your time and test it.
Those are just some messages for printing the configured orientation, are for debug info, i will remove them.

Thanks again, i think we can close this issue :)

PR #114 (2020-05-06): Ili9486 rotation
Closes #111
closed 2020-05-06 by C47D