lvgl/lv_esp_idf #112: TFT SPI queued but never dequeued - can lead to potential memory leaks

davidjade, 2020-05-05:
Sorry for the long narrative but this doesn't fit the bug template as this is a general design issue with the current disp_spi driver.

I've been working with the lvgl esp32 driver code (adding a new display type actually) and something strikes me as wrong.

Code in the TFT disp_spi calls spi_device_queue_trans() but it never calls spi_device_get_trans_result(). From reading the IDF spi_master source, this places items into an xQueue for received transactions but leaves them there. This causes the spi_master received queue to overflow silently because spi_master does not check for that. I guess if you never need results, this is maybe somewhat Ok. However, more worrisome is that sometimes the spi_master will allocate and attach DMA memory to a transaction (if the transaction buffers passed in are in non DMA-able memory) and this only gets freed if you retrieve the transaction results (via uninstall_priv_desc()).

There is a note in spi_master.c about this (in spi_device_get_trans_result(), around line 935): 
```
// The memory occupied by rx and tx DMA buffer destroyed only when receiving from the queue (transaction finished).
```
So if you have conditions that cause spi_master to allocate DMA memory and you never retrieve the SPI transaction results, then that memory never gets freed.

Maybe no one has seen this since a lot of folks statically allocate the display buffers and that ends up in IRAM. In my case I have a 800x480 16-bit screen device and I trying to see how it would perform using PSRAM in some cases, which is not DMA-able as far as I know. With the current lvgl_esp32_driver design, these internally allocation spi_master DMA buffers will never get freed and will lead to memory leaks.

I think the prudent thing to do is to always wait for and retrieve transaction results before sending new transactions, so that both the spi_master receive xQueue doesn't overflow and any internally attached DMA memory gets freed. This likely will cause how spi_trans_in_progress is signaled to have to change too as it would then be more complicated than just  post_cb being called as a transaction really isn't finished until the results are retrieved.

C47D, 2020-05-06:
Hi @davidjade thanks for reaching out.

> Code in the TFT disp_spi calls spi_device_queue_trans() but it never calls spi_device_get_trans_result(). From reading the IDF spi_master source, this places items into an xQueue for received [...]

Just saw the esp-idf source and your analisys seems to be correct, I'm not an esp-idf expert tho. I guess that we don't notice the xQueue being overflowed because we constantly draw the display.
As far as I know the only driver (in master branch) that allocates DMA memory is the ILI9488 (it got updated recently #109), and now that you mention it maybe their problem is related to the memory not being destroyed.

I will try to use the ILI9488 display and assert the return values of the spi_device_queue_trans.

> Maybe no one has seen this since a lot of folks statically allocate the display buffers and that ends up in IRAM. In my case I have a 800x480 16-bit screen device and I trying to see how it would [...]

Browsing the [esp_spi_master_example](https://github.com/espressif/esp-idf/blob/c9f29e0b59988779bf0792e2e954c3448fd37e3b/examples/peripherals/spi_master/lcd/main/spi_master_example_main.c#L364), I see that they call the `spi_device_get_trans_result` function in a loop, do you think it is possible to call that function on the interrupt triggered by the SPI?

Currently it looks like this (in master):

```c
static void IRAM_ATTR spi_ready (spi_transaction_t *trans)
{
    spi_trans_in_progress = false;

    lv_disp_t * disp = lv_refr_get_disp_refreshing();
    if(spi_color_sent) lv_disp_flush_ready(&disp->driver);
    if(chained_post_cb) chained_post_cb(trans);
}
```

Now the big question.
Should we fix this particular problem on the master branch or should we fix it on the LVGL v7 port pull request, maybe in both?

See #110 for my toughts on your other suggestions.

davidjade, 2020-05-06:
One thing to note is that spi_master may allocate DMA memory on your behalf even if you didn't explicitly do this - it decides. This is why transactions always need to be cleared I think. I do think it is unlikely that spi_master would do this in typical lvgl cases (or you'd see many more bug reports I think).

It is not possible to call spi_device_get_trans_result in the ISR, it will cause kernal panics. It has to be done outside of the ISR. 

I have a working version that does that and what I do is keep track of pending queued transactions and clear them before starting new transactions. This handles a mix of polling and queued transactions. It gets a little more complicated because to get a transaction result, you need to keep the original transaction that was sent in memory (results are just an updated version of that memory). But I do have something that works. It's a work in progress as I hack away at this but I could potentially clean it up at some point. Just note that I took a re-write vs. tweak approach. What I ended up with is something like this:

``` C

typedef enum _spi_send_flag_t
{
    SPI_SEND_QUEUED		= 0x00000000,
    SPI_SEND_POLLING		= 0x00000001,
    SPI_SEND_SYNCHRONOUS	= 0x00000002,
    SPI_SEND_SIGNAL_FLUSH	= 0x00000004
} spi_send_flag_t;

void disp_spi_send_data_ex(uint8_t * data, uint16_t length, spi_send_flag_t flags)
{
    if (length == 0) return;           //no need to send anything

    // retrieve pending previous transaction results
    static volatile uint8_t pending = 0;
    spi_transaction_t* presult;
    while(pending)
    {
        if(spi_device_get_trans_result(spi, &presult, portMAX_DELAY) == ESP_OK) {
            pending--;
        }
    }

    static spi_transaction_t t;		// must hang around until retrieval
    memset(&t, 0, sizeof(t));

    if (length <= 4) {
        t.length = length * 8;
        t.flags = SPI_TRANS_USE_TXDATA;
        memcpy(t.tx_data, data, length);
    } else {
        t.length = length * 8; // transaction length is in bits
        t.tx_buffer = data;
    }

    t.user = (void*)flags;		// save flags for pre/post transaction processing

    // poll/complete/queue transaction
    if(flags & SPI_SEND_POLLING) {
        spi_device_polling_transmit(spi, &t);
    } else if (flags & SPI_SEND_SYNCHRONOUS) {
        spi_device_transmit(spi, &t);
    } else {
        pending++;
        spi_device_queue_trans(spi, &t, portMAX_DELAY);
    }
}

static void IRAM_ATTR spi_ready (spi_transaction_t *trans)
{
    spi_send_flag_t flags = (spi_send_flag_t)trans->user;
    if(flags & SPI_SEND_SIGNAL_FLUSH) {
        lv_disp_t * disp = lv_refr_get_disp_refreshing();
        lv_disp_flush_ready(&disp->driver);
    }
    if(chained_post_cb) chained_post_cb(trans);
}

```

A few notes:

- This currently has a bug in that a non-queued transaction will overwrite a queued transaction. The fix is to only have queued transactions be static/saved - I will fix that later today.
- changed to signaling flush complete per-transaction so that if I need to queue multiple transactions but only signal flush complete on the last one, I can do that. I use the user field to handle this.
- pending is a counter because I may need to use more than one queued item in my driver so that's just me planning for allowing more than one transaction, etc.. This will require me to keep track of all pending transaction though.
- to maintain compatibility, disp_spi_send_data and disp_spi_send_colors now just call this one SPI send function using the appropriate flags.
- last, this is not fully tested and very much a work in progress.

One other thing I am going to investigate soon is, using the CMD and ADDRESS capabilities of SPI transactions. My display is a memory-mapped device that needs addresses sent as part of the data transaction (i.e. same CS line cycle) and if I can use those fields successfully, that I might be able to let the driver control CS and do away with my current split transaction need. This will require more flags on sending, etc... Another approach might be to make the ISR control CS via a flag like it does for flushing.

I'm not opposed to working it into either v6 or v7 but I'm not ready to switch to v7 at this point. I'd like to stay with something stable as I continue to work on Quad SPI, CS control line work, etc... I have not been tracking lvgl 7 closely though so I don't know how stable/unstable it is.

I'll also profess to not being a ESP expert either, this is just what I have figured out so far.

C47D, 2020-05-07:
Thanks for the code snippet, I've added it in here: [rework_display_spi](https://github.com/littlevgl/lv_port_esp32/tree/rework_display_spi), that branch is based on master (LVGL v6), I'm planning to add it to the LVGL v7 PR once we got it working. I'm running tests with a ILI9488 display.

> One other thing I am going to investigate soon is, using the CMD and ADDRESS capabilities of SPI transactions. My display is a memory-mapped device that needs addresses sent as part of the data [...]

On the LVGL v7 PR, the only functions on `disp_spi` are:
https://github.com/littlevgl/lv_port_esp32/blob/d896b931809641ac1afa03ae11355d6923bde8e2/components/lvgl_esp32_drivers/lvgl_tft/disp_spi.c#L55-L76

Currently being used like this:
https://github.com/littlevgl/lv_port_esp32/blob/d896b931809641ac1afa03ae11355d6923bde8e2/components/lvgl_esp32_drivers/lvgl_driver.c#L48-L63

On `disp_spi_add_device` we could add parameters to configure the `command_bits` and `address_bits` of the `spi_device_interface_config_t` and default them to 0. `lvgl_driver_init` is just a help function so the end users doesn't have to configure the spi or i2c buses, you can do the same on your application.

> I'm not opposed to working it into either v6 or v7 but I'm not ready to switch to v7 at this point. I'd like to stay with something stable as I continue to work on Quad SPI, CS control line work, etc... I have not been tracking lvgl 7 closely though so I don't know how stable/unstable it is.

I would like to left the current master branch working with all the improvements on the display spi code, maybe adding some flags to the spi bus configuration to let you enable the QSPI pins.

About the CS control, I would like to the code control it, afaik hardware control have some restrictions.

Let me know if I'm missing something on the new branch, thanks for the suggestions, I would love to have some more free time to work on this, but I have not :(.

- Carlos

davidjade, 2020-05-07:
Just a quick coupled of notes, since I was just working on this. I will take a look at your new code soon.

For my display, I cannot use CMD and ADDR bits at the device config level - I have to use them on the transaction level since not all of my transactions need address bits. This has cause my re-work of disp_spi_transaction_ex (of whatever I end up calling it) to use spi_transaction_ext_t and additional flags to control when to use address bits. I have also re-worked it to support reading data, another requirement I have. So now I have on universal SPI transaction function that meet all my needs.

For the CS line requirement I had, I have worked around that requirement by building more complete transaction that including both reading, writting, and address bits. This is possible because my display is a full duplex device (another thing in device setup I had to allow for). So I don't think I will manual need control over CS but I am still working towards that so I don't know for sure yet.

At this point my SPI code is fairly full-featured as it allows per-transaction writing, reading, address bits (only 24-bit right now), etc... 

Here's what my code look like right now (still not complete but getting there):

```c 
typedef enum _spi_send_flag_t
{
	SPI_SEND_QUEUED			= 0x00000000,
	SPI_SEND_POLLING			= 0x00000001,
	SPI_SEND_SYNCHRONOUS	= 0x00000002,
	SPI_SIGNAL_FLUSH			= 0x00000004,
	SPI_RECEIVE				= 0x00000008,
	SPI_ADDRESS_24			= 0x00000100,
} spi_send_flag_t;

typedef struct _spi_read_data
{
	uint8_t _dummy_byte;
	union
	{
		uint8_t 	byte;
		uint16_t	word;
		uint32_t	dword;
	} __attribute__((packed));
} spi_read_data __attribute__((aligned(4)));

void disp_spi_transaction_ex(const uint8_t* data, uint16_t length, spi_send_flag_t flags, spi_read_data* out, uint64_t addr)
{
    if (length == 0) return;           //no need to send anything

	// clear previous pending transaction results
	disp_clear_pending_transactions();

	spi_transaction_ext_t t = {0};

	t.base.length = length * 8; // transaction length is in bits

	if (length <= 4 && data != NULL) 
	{
		t.base.flags = SPI_TRANS_USE_TXDATA;
		memcpy(t.base.tx_data, data, length);
	} 
	else 
	{
        	t.base.tx_buffer = data;
	}

	if(flags & SPI_RECEIVE) 
	{
		assert(out !=NULL && (flags & (SPI_SEND_POLLING | SPI_SEND_SYNCHRONOUS)));	// sanity check
		t.base.rx_buffer = out;
		t.base.rxlength = 0;	// default to same as tx length
	}

	if(flags & SPI_ADDRESS_24) 
	{
		t.address_bits = 24;
		t.base.addr = addr;
		t.base.flags |= SPI_TRANS_VARIABLE_ADDR;
	}

	t.base.user = (void*)flags;		// save flags for pre/post transaction processing

	// poll/complete/queue transaction
	if(flags & SPI_SEND_POLLING) 
	{
		spi_device_polling_transmit(spi, (spi_transaction_t*)&t);
	} 
	else if (flags & SPI_SEND_SYNCHRONOUS) 
	{
		spi_device_transmit(spi, (spi_transaction_t*)&t);
	} 
	else 
	{
		static spi_transaction_ext_t queuedt;		// must hang around until retrieval
		memcpy(&queuedt, &t, sizeof(t));
		spi_pending_trans++;
		spi_device_queue_trans(spi, (spi_transaction_t*)&queuedt, portMAX_DELAY);
	}
}

bool disp_spi_is_busy(void)
{
    return spi_pending_trans > 0;
}

void disp_clear_pending_transactions(void)
{
	spi_transaction_t* presult;
	while(spi_pending_trans)
	{
		if(spi_device_get_trans_result(spi, &presult, portMAX_DELAY) == ESP_OK) {
			spi_pending_trans--;
		}
	}
}

bool disp_spi_is_busy(void)
{
    return spi_pending_trans > 0;
}

void disp_clear_pending_transactions(void)
{
	spi_transaction_t* presult;
	while(spi_pending_trans)
	{
		if(spi_device_get_trans_result(spi, &presult, portMAX_DELAY) == ESP_OK) {
			spi_pending_trans--;
		}
	}
}
```
Notes:
- my read code always sends a dummy byte as that is what my device needs. That seems pretty typical but it should probably be a configurable option. 
- due to the need to clear transactions, some others driver will need to call disp_clear_pending_transactions rather than just loop on disp_spi_is_busy. In fact, I think disp_spi_is_busy should just be removed and waiting/loop for SPI should be left to the driver to handle.

C47D, 2020-05-09:
Hi,

With 5d03411 I've added parameters to the `lvgl_spi_driver_init` function to set the QSPI pins on the SPI master.

> due to the need to clear transactions, some others driver will need to call disp_clear_pending_transactions rather than just loop on disp_spi_is_busy. In fact, I think disp_spi_is_busy should just be removed and waiting/loop for SPI should be left to the driver to handle.

Do you mean on here?
```c
void disp_spi_send_data(uint8_t * data, uint16_t length)
{
    if (length == 0) return;           //no need to send anything

    while(spi_trans_in_progress); // Busy waiting SPI to finish
    
    spi_transaction_t t = {0};

    if (4 > length) {
        t.length = length * 8;
        t.flags = SPI_TRANS_USE_TXDATA;
        memcpy(t.tx_data, data, length);
    } else {
        t.length = length * 8; // transaction length is in bits
        t.tx_buffer = data;
    }

    spi_trans_in_progress = true;
    spi_color_sent = false;             //Mark the "lv_flush_ready" NOT needs to be called in "spi_ready"
    spi_device_polling_transmit(spi, &t);
}
```

I'm a bit confused on how ESP32 SPI works, hope you or anyone else can help me, if we configure the SPI master with more than 1 queue depth, could we queue the transaction instead of waiting the SPI to be done with it's current transaction?

Once it finished the current transaction, the SPI by itself gets the next transaction and send it over the wire?

Once the transaction queue gets empty, the interrupt is generated?

davidjade, 2020-05-10:
To clear transaction results, you have to call code like this (from my most recent revised code I pasted in above):

``` c
void disp_clear_pending_transactions(void)
{
	spi_transaction_t* presult;
	while(spi_pending_trans)
	{
		if(spi_device_get_trans_result(spi, &presult, portMAX_DELAY) == ESP_OK) {
			spi_pending_trans--;
		}
	}
}
```

The important part is calling spi_device_get_trans_result(). It has to be one-for-one with transactions sent or you will not release any memory that the ESP32 spi_master might have allocated for those transactions. Note that the pointer to the transaction it returns is the same transaction structure that was submitted earlier, so it must be kept in memory somewhere (I statically allocate one).

My last example uses code like this instead of spinning on the current spi_trans_in_progress (which I removed). It serves the same purpose, to ensure that the data has arrived. 

If you used multiple queued transactions and you needed to know when all the data arrived, you still need to do the same. Some of these displays which need to toggle data vs cmd line would have to use a wait function like this too to ensure that all the transaction have actually arrived - i.e. the transaction is complete. i.e. you have to keep track of them all.

Also if you have more than one transaction, the ESP32 will send them in sequence but you still have to get the results for each one at some point and each one will signal via an interrupt. So the logic for handling lvg flushing and retrieving transactions results with minimal blocking gets more complicated. My code does not handle that yet (but it will at some point). Note also that if you have more than one transaction in play, then you need multiple allocated (or static) transactions structures that stick around for the duration so that results can be retrieved. Retrieving results effectively returns a reference to the transaction structure that was sent - it seems it has to be kept in memory until results are retrieved

For Quad SPI, I don't know how it works for other devices but for the display I am using it appears that it first needs to talk using 2-wire SPI, tell it to transition to Quad SPI, then it starts talking via Quad SPI. This requires tearing down the SPI channel (or re-configuring) it to switch between 2-wire and 4-wire modes. So this will also get more complicated than set and forget SPI setup. There is no way for it to know the master is using Quad mode otherwise from that I can tell.

My current plan is to clean up what I have for lvg 6 and the current master EPS32 drivers for 2-wire mode. It supports a full featured SPI that handles transactions, clearing transaction results, reading, writing, etc... I can them submit that via a pull request on a new branch. If it's useful then I am happy to submit it.

I'll also submit my new driver for the EVE2 line of displays. It is a lot more complicated that the other drivers which is why it needed such a revised SPI driver layer. 

Sorry I have not looked at the new code much or the direction it is taking as I really needed to start with something stable.

Down the road I will port what I have to lvg 7 if that makes sense but right now I don't have the bandwidth to track a moving target.

closed 2020-06-09 by stale[bot]