lvgl/lv_esp_idf #115: ESP get started page

kisvegabor, 2020-05-07:
Hi,

In the docs of v7 there will be sites for manufacturers and platforms.
For ESP it is here: https://docs.littlevgl.com/v7/en/html/get-started/espressif.html

It's similar to a README. A few lines about ESP, some recommended boards, how to get started, some platform-specific info, a super simple example, etc.

I'd like to ask if somebody has time and energy to write about these things.

C47D, 2020-05-09:
I would love to do that but I'm pretty busy right now. Have you posted this on the forum?

kisvegabor, 2020-05-11:
No problem!

> Have you posted this on the forum?
Not yet, I wanted to ask here first.

Thank you anyway! :slightly_smiling_face:

closed 2020-05-11 by kisvegabor