lvgl/lv_esp_idf #117: Is it supported for "esp-idf v4.2" branch?

guo652917087, 2020-05-08:
Hi,
   When I use "esp-idf v4.2" and use the command "idf.py menuconfig", the menu function cannot be run successfully. I can switch to v4.0 and it will work fine. Is there any plan to support v4.2 version?

C47D, 2020-05-08:
Hi,

I haven't tested the ESP-IDF 4.2 branch, I'm running ESP-IDF 4.1-rc1 myself. Care to send an screenshot? Do it shows something on the console?

guo652917087, 2020-05-08:
Hi，
I just tested it again and found that the menu can be generated on linux, and the menu cannot be generated on windows10, and an error is reported. It should be an error in the CMakeLists.txt file.

So I chose to compile with linux.

[image]

closed 2020-05-08 by guo652917087

C47D, 2020-05-08:
Seems like CMake can't find the C nor C++ compilers, are the esp32 dependencies installed? I think each release of the ESP-IDF need to install their dependencies.