lvgl/lv_esp_idf #118: Invert display option name misleading

rangeli, 2020-05-08:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**
The option "Invert display" is very misleading.  "Invert display" sounds like the colors are being inverted when in reality, the image is being **flipped**.

I suggest that we change the name for this option to "Flip display", but I don't know if this might break compatibility with projects that use this library.

**Code to reproduce the issue**

**Expected Results**

**Actual Results**

**ESP32 Chip version**

**ESP-IDF version**

**Development kit used**

**Development machine OS**

**Compilation warnings/errors (if available)**

**If possible, copy the compilation log into a file and attach it here**

C47D, 2020-05-08:
Hi @rangeli,

I agree with you, we've been adding display rotation to some display controllers and noticed the invert display option intended to do that. So what do you think if we remove the invert display and left the display rotation option only?

Just out of curiosity, what display controller are you using?

C47D, 2020-05-09:
@rangeli Is this closed with #121?

closed 2020-05-31 by C47D