lvgl/lv_esp_idf #119: Colors are inverted on ILI9341

rangeli, 2020-05-08:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Describe the issue**
I flashed the example to my [M5Stack Gray](https://docs.m5stack.com/#/en/core/gray) and the colors were inverted.

I tried using the "Invert display" feature, but found that it actually flipped the display instead.

I was able to fix this by sending an invert display command to the ILI9341, but I had to modify the code to do this.

I've created a PR with the change, but I'm not sure if this issue only affects some M5Stack models.

**Code to reproduce the issue**

**Expected Results**

**Actual Results**

**ESP32 Chip version**

**ESP-IDF version**

**Development kit used**

**Development machine OS**

**Compilation warnings/errors (if available)**

**If possible, copy the compilation log into a file and attach it here**

C47D, 2020-05-08:
Hi,

I don't have that dev kit, can you take a pic of the inverted colors? IIRC I had the same problem but with other display.

Based on your other issue, we may add an option to the display menu config named "Invert display colors" or something similar, so the users can enable/disable at will. What do you think?

rangeli, 2020-05-08:
I think having a menu item for this might make sense, at least for the M5Stack/ILI9341.  I've made a PR with the fix BTW, but I could turn it into a menuconfig option.  Here's some pics of the display before and after the fix:
[image]

[image]

C47D, 2020-05-08:
Thanks for the photo and PR, the problem can be specific to the M5Stack, I'm using the Wrover kit for my tests and I didn't got the inverted colors.

By any chance are you configuring the display orientation to any of the inverted options?

barbiani, 2020-05-08:
Try to change your lv_conf.h to `#define LV_COLOR_16_SWAP   1`

rangeli, 2020-05-09:
> Try changing your lv_conf.h to `#define LV_COLOR_16_SWAP 1`

Unfortunately, that doesn't work.  That option swaps the the 2 bytes of RGB565 color.  The fix needs to be a full invert of the colors.

rangeli, 2020-05-09:
I've added a menuconfig option for this:

[image]

@C47D Can you look at my PR?

C47D, 2020-05-09:
@rangeli Is this closed with #121 ?

closed 2020-05-23 by C47D