lvgl/lv_esp_idf #127: Is it possible to read the ID of the screen chip?

guo652917087, 2020-05-09:
Hi，
   Is it possible to read the ID of the screen chip? No read function is defined.
    I use ST7789V2
    Why does ST7789V use SPI mode 2?
   
   I modified and modified st7789.c to increase the vertical screen display:

   {ST7789_MADCTL, {0x80|0x20}, 1}, // Set to 0x28 if your display is flipped

C47D, 2020-05-09:
> Is it possible to read the ID of the screen chip? No read function is defined.

Right now the display river only uses the MOSI signal, what register do you need to read to get the ID?

> I modified and modified st7789.c to increase the vertical screen display:
{ST7789_MADCTL, {0x80|0x20}, 1}, // Set to 0x28 if your display is flipped

Thanks for the code snippet, I will add more display orientation options to the ST7789, I got the IPS 240x240 display.

I had been busy updating the code for LVGL v7 and ILI9341 and ILI9488 displays, will work on the ST7789 now that I know you are using it.

guo652917087, 2020-05-11:
@C47D 
Right now the display river only uses the MOSI signal, what register do you need to read to get the ID?
Reply：Now the screen display function is written directly. Can you consider that the "littlevgl" display will have a flower screen (the display is disordered), if I regularly (for example, 3 minutes) compare the content written in and the content read out is the same? Just to be more stable. My idea is that if you read the ST7789V ID (non-zero value), if the reading fails, the chip works abnormally, you need to reset ST7789V and reinitialize.
Because there is no reading function, even if my screen is removed, the program can be executed down.
I don't remember to be honest, it worked for me and I leave it like that, it doesn't work for you?
Reply：It works. I use  ST7789V2   (240*320)

guo652917087, 2020-05-20:
No feedback, then I close this issue.

closed 2020-05-20 by guo652917087