lvgl/lv_esp_idf #128: idf.py failed in windows7

ghosttk, 2020-05-12:
Build without modifing anything results in the following:
`
../components/lvgl_esp32_drivers/lvgl_touch/ft6x36.c:23:10: fatal error: lvgl/lv
gl.h: No such file or directory
 #include <lvgl/lvgl.h>
          ^~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
`

I  copy lvgl.h from littlgvgl/lvgl repository to lvgl/lvgl directory and rebuilt, but resulting other "no such file" errors.

C47D, 2020-05-12:
Hi @ghosttk,

What display and touch controller are being used in your configuration? I need more information to try replicate the issue.

ghosttk, 2020-05-12:
I did not change the dufault configuratonns, it was still 9341.

C47D, 2020-05-12:
How did you cloned the repo? Are the submodules initialized? You can take a look at components/lvgl/lvgl, if there is no files in that directory then you need to initialize the submodules.

I have no Windows 7 machines available but I just cloned the repo and was able to build the project with default configuration.

ghosttk, 2020-05-13:
Problem solved ,  thank you!!
It was because I used to directly download repositories from browser, hence not including submodules.

closed 2020-05-13 by ghosttk

C47D, 2020-05-13:
Don't worry, have fun :)