lvgl/lv_esp_idf #132: Driver for ST7796S

LuoJuly, 2020-05-21:
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

**Can you add st7796s driver ? Thank you sir**

**He seems to be able to use the ili9341 driver, but he flashes every few seconds**

**ESP32 Chip version：ESP32-WROVER-B+ST7796S 320*480 3.5inch LCD SCREEN**

LuoJuly, 2020-05-21:
[image]

C47D, 2020-05-21:
Hi,

I will take a look at the datasheet of it, I don't have any display with that driver at hand.
What version of ESP-IDF are you using?

closed 2020-06-20 by stale[bot]