lvgl/lv_esp_idf #133: Can't get it to work with ESP-IDF

henkjannl, 2020-05-21:
After:
```
git clone --recurse-submodules https://github.com/littlevgl/lv_port_esp32.git
cd lv_port_esp32
%userprofile%\esp\esp-idf\export.bat
idf.py menuconfig
```

I get:
```
...
...
- Check for working CXX compiler: C:/Users/hjvanderpol/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Users/hjvanderpol/.espressif/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at C:/Users/hjvanderpol/esp/lv_port_esp32/main/CMakeLists.txt:2 (idf_component_register):
  Unknown CMake command "idf_component_register".
Call Stack (most recent call first):
  C:/Users/hjvanderpol/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:107 (include)
  C:/Users/hjvanderpol/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:217 (expand_component_requirements)

CMake Error at C:/Users/hjvanderpol/esp/esp-idf/tools/cmake/idf_functions.cmake:338 (message):
  Failed to expand component requirements
Call Stack (most recent call first):
  C:/Users/hjvanderpol/esp/esp-idf/tools/cmake/project.cmake:173 (idf_import_components)
  CMakeLists.txt:7 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/hjvanderpol/esp/lv_port_esp32/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/hjvanderpol/esp/lv_port_esp32/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

```
My directory structure looks like this:
```
...esp
    esp-idf
    lv_port_esp32
    hello_world
```

The hello_world example is working normally

Cmakelist.txt just contains:
```
...
5> if (NOT DEFINED PROJECT_NAME)
6> 	project(lvgl-demo)
7> endif (NOT DEFINED PROJECT_NAME)

```

Just ignoring that and running idf.py build also produces errors.

What am I doing wrong?

C47D, 2020-05-21:
Hi,

I tried to replicate your issue with an Ubuntu machine, but I didn't had any issue, so I guess this issue is specific to Windows.

Can you try using the `ESP-IDF Command Prompt` created when you installed all the tools?

I will try to get a Windows machine to replicate the issue later today.

Regards
Carlos

C47D, 2020-05-21:
I'm able to make it work with the ESP-IDF Command Prompt.

[image]

henkjannl, 2020-05-21:
Dear Carlos,

Thanks for helping. I was already using the ESP-IDF command prompt. I'm not
sure which version of ESP-IDF I'm using but I think it is 3.3. Also, Python
3 is installed, could it be something like that?

HenkJan

C47D, 2020-05-21:
I'm using ESP-IDF 4.0, can you try that one? I'm 99% sure is because of that.

henkjannl, 2020-05-22:
Dear Carlos,

That indeed works! I'm working today, so did not have time to test all, but
menuconfig works and build works without errors.

Thank you very much!

Would it be a suggestion to add something in the README.md? It
currently says:

> This project is compatible with both the ESP-IDF 3.X branch and the 4.0
> branch. For the 4.0 branch, you can use either make or CMake, but CMake is
> recommended. Instructions here are given for the latest toolchain using
> idf.py (which invokes CMake), but it is easy to translate to make. For
> example instead of running idf.py menuconfig, just run make menuconfig.

I'm sure "is easy translate to make" is true if you know what you are
doing. My case is that I want to migrate from Arduino IDE to something more
advanced, and I'm initially just looking for a robust ILI9341, driver
(Bodmer depends on Arduino, Loboris seems outdated).

Thanks again,

HenkJan van der Pol.

closed 2020-05-22 by henkjannl